rvdbreemen/OTGW-firmware #149: MQTT broker not connected

marcoCasamento, 2022-11-30:
Hi and thanks for the great FW.
I use the firmware on a Nodosvop v2 with a PIC16F88 on PIC FW 5.0.1 and MCU FW 0.9.5+4eb7d24 (30-05-2022).
It's configured on a MQTT server that I run in a container and seems to works just fine, but I've noticed that HomeAssistant about every 10/15 minutes say that all entities provided by the MQTT integrations are unavailable.
If I connect through telnet on the MCU I see:

```
15:05:07.138281 (  12656| 11544) processOT   (1693): Answer Thermostat AC0000300 (9)[MsgID=  0][READ_ACK        ]>Status = Slave  [--------]
15:05:07.174146 (  11312| 10248) processOT   (1693): Thermostat        T00000300 (9)[MsgID=  0][READ_DATA       ]-Status = Master [CD---W--] <ignored>
15:05:17.409951 (  10016|  8832) processOT   (1693): Request Boiler    R80000100 (9)[MsgID=  0][READ_DATA       ]>Status = Master [C----W--]
15:05:17.420603 (  10264|  8832) handleOTGW  (1760): Serial Overrun
15:05:17.422506 (   8920|  8184) processOT   (1722): Not processed, received from OTGW => (0300) [4]
15:05:17.434596 (   9320|  8184) sendMQTTData( 406): Error: MQTT broker not connected.
15:05:17.437355 (   7976|  6888) sendMQTTData( 406): Error: MQTT broker not connected.
15:05:17.438049 (   7976|  6888) sendMQTTData( 406): Error: MQTT broker not connected.
15:05:17.438485 (   7976|  6888) sendMQTTData( 406): Error: MQTT broker not connected.
15:05:17.438929 (   7976|  6888) sendMQTTData( 406): Error: MQTT broker not connected.
15:05:17.439365 (   7976|  6888) sendMQTTData( 406): Error: MQTT broker not connected.
15:05:17.439809 (   7976|  6888) sendMQTTData( 406): Error: MQTT broker not connected.
15:05:17.440904 (   7304|  6240) sendMQTTData( 406): Error: MQTT broker not connected.
15:05:17.441753 (   7304|  6240) sendMQTTData( 406): Error: MQTT broker not connected.
15:05:17.442248 (   7304|  6240) sendMQTTData( 406): Error: MQTT broker not connected.
15:05:17.442746 (   7304|  6240) processOT   (1693): Boiler            B40000100 (9)[MsgID=  0][READ_ACK        ]>Status = Slave  [--------]
15:05:17.450384 (   9320|  8184) sendMQTTData( 406): Error: MQTT broker not connected.
```
What can be the problem ? How can I troubleshoot it ?

rvdbreemen, 2022-12-20:
Somehow it looks like the wire protocol is broken for some reason in your case. I have not seen this myself, but I will go and look in my MQTT logs to see if that is really the case. Has you situation resolved it self? Or still seeing this strange behaviour?

marcoCasamento, 2022-12-20:
Yes, sorry, this must be closed!

 

I’ve had the same doubt that there were some connection issues. 

I’ve put a ping running for a few hours and it reported less than 1% of loss packets, but still SOME of them were losts. I’ve changed the ESP board with no luck, then I’ve tried to change firmware and I had the same problem.

 

Finally, last weekend I’ve moved the container running mosquitto to a different server and I had 24+ hours without problem! So the problem is NOT in the firmware and this must be closed. Sorry, I should have put an update on the issue.

 

Just yesterday evening I’ve also discovered that my linux server was using two ip addresses (one static and one by dhcp client). Removing the address from dhcp for some unknown reason put the lost packet to 0.

rvdbreemen, 2022-12-21:
Thanks for letting me know.

closed 2022-12-21 by rvdbreemen