rvdbreemen/OTGW-firmware #150: MQTT values sometimes mixed up

chokays, 2022-12-08:
From the log:

12:24:09.107921  T001A7FFF  Read-Data   DHW temperature (MsgID=26): 128.00
12:24:09.282683  B401A1520  Read-Ack    DHW temperature (MsgID=26): 21.13
12:24:09.599133  T90383700  Write-Data  DHW setpoint (MsgID=56): 55.00
12:24:09.900696  B50383700  Write-Ack   DHW setpoint (MsgID=56): 55.00
12:24:10.105121  T00197FFF  Read-Data   Boiler water temperature (MsgID=25): 128.00
12:24:10.311940  Error 04
12:24:10.314347  EC01118E0
12:24:10.597268  T00197FFF  Read-Data   Boiler water temperature (MsgID=25): 128.00
12:24:10.919690  BC01918E0  Read-Ack    Boiler water temperature (MsgID=25): 24.88
12:24:11.108307  T00110000  Read-Data   Relative modulation level (MsgID=17): 0.00
12:24:11.321138  BC0110000  Read-Ack    Relative modulation level (MsgID=17): 0.00

[image]

The MQTT value in HA for rel mod level is 24.88 as shown in the image, but the log shows that 24.88 is boiler water temp.

Another example:

12:28:02.776816  T90383700  Write-Data  DHW setpoint (MsgID=56): 55.00
12:28:03.002427  B50383700  Write-Ack   DHW setpoint (MsgID=56): 55.00
12:28:03.264088  T00197FFF  Read-Data   Boiler water temperature (MsgID=25): 128.00
12:28:03.412253  Error 04
12:28:03.414725  EC0111740
12:28:03.771522  T00197FFF  Read-Data   Boiler water temperature (MsgID=25): 128.00
12:28:04.024926  BC0191740  Read-Ack    Boiler water temperature (MsgID=25): 23.25
12:28:04.258964  T00110000  Read-Data   Relative modulation level (MsgID=17): 0.00
12:28:04.419579  BC0110000  Read-Ack    Relative modulation level (MsgID=17): 0.00

[image]

Another example:

12:47:31.589928  BC0000102  Read-Ack    Status (MsgID=0): 00000001 00000010
12:47:31.763482  T001A7FFF  Read-Data   DHW temperature (MsgID=26): 128.00
12:47:31.999438  BC01A1420  Read-Ack    DHW temperature (MsgID=26): 20.13
12:47:32.268363  T90383700  Write-Data  DHW setpoint (MsgID=56): 55.00
12:47:32.393784  B50383700  Write-Ack   DHW setpoint (MsgID=56): 55.00
12:47:32.774861  T00197FFF  Read-Data   Boiler water temperature (MsgID=25): 128.00
12:47:33.009781  Error 04
12:47:33.012587  E40111640
12:47:33.263710  T00197FFF  Read-Data   Boiler water temperature (MsgID=25): 128.00
12:47:33.421710  B40191640  Read-Ack    Boiler water temperature (MsgID=25): 22.25
12:47:33.771540  T00110000  Read-Data   Relative modulation level (MsgID=17): 0.00
12:47:34.021796  BC0110000  Read-Ack    Relative modulation level (MsgID=17): 0.00
[image]

Seems to be related to the Error 4 just before these occurrences?

rvdbreemen, 2022-12-20:
I will have to look into this, is this still happening?

chokays, 2023-01-25:
I'm still having the issue

rvdbreemen, 2023-01-26:
It seems that this is cause by decoding and sending messages with parity error. Just fixed this in dev branch. If you are on Discord community channel, you can help test this by using the beta. So please join: https://discord.gg/mSQBYBMNRy

closed 2023-01-26 by rvdbreemen