rvdbreemen/OTGW-firmware #151: Wemos D1 not working after flash

poky0611, 2022-12-10:
Hello,

I flashed Wemos D1 mini and after flash I have only this on serial output:  "wrong chip id 0xf4fc"

Where can be problem? I flashed it with this command: "C:\Users\ropo\AppData\Local\Arduino15\packages\esp32\tools\esptool_py\4.2.1/esptool.exe" --port "COM3" --baud 921600  --before default_reset --after hard_reset write_flash  -z --flash_mode dio --flash_freq 80m --flash_size 4MB 0x0 "ino.bin" 0x200000 "fs.bin"

Can you help me please?
Thank You!
Robert Pokorný

rvdbreemen, 2022-12-20:
Hi Robert, Thanks for reporting this. Not sure myself what's going on, but it seems that there is some issue with support on the current flashtools. I will have to look into this to see if I can replicate this.

0crap, 2022-12-20:
On Windows flashing the Wemos D1 went fine with the flash_download_tool_3.8.5.exe
However this version is not for download any more on the [official website](https://www.espressif.com/en/support/download/other-tools).

I tried the latest version from the above website, v3.9.3.
It fails every time without a clear error message why.

Just my 2ct.

tjfsteele, 2022-12-22:
You can still get 3.8.5 from this link. I've put it in the wiki.

https://drive.google.com/file/d/1_M4EzolaJWpYXts_FwUIqH8pZWqy-fye/view?usp=sharing

0crap, 2023-01-14:
Seems to be a duplicate of #135

rvdbreemen, 2023-01-14:
You are correct, the issue seems to be that the flashtools of later versions are not working correctly. I will try myself on my wemos unit.

rvdbreemen, 2023-01-14:
Closing this issue as it has my attention, and the next release can be flashed once more with the UI tool from Espressive.

closed 2023-01-14 by rvdbreemen