rvdbreemen/OTGW-firmware #152: Add I2C support for OLED/LCD

STemplar, 2022-12-11:
Add I2C support for LCD 16x2/20x4 and/or OLED Display with I2C Modules as suggested by the OpenTherm Gateway Manual.
[image]

STemplar, 2022-12-21:
Well, I am not a programmer. But if it's a standard Arduino library, I can take a look and give it a try.

Roos-AID, 2023-01-17:
I2C is already used for temperature sensors, so not a big deal to add some function. However, it requires quite some memory for display rendering.  They key question here is what you would like to display on the LCD  and what benefit would that have over HA ?

STemplar, 2023-01-22:
> 

As far as I know the [Arduino I2C LCD](https://arduinogetstarted.com/tutorials/arduino-lcd-i2c) interface looks very much like a serial interface. No display rendering, so perhaps a simple LCD-displey is more suited than an OLED-display.

DaveDavenport, 2023-01-22:
u8g2 should be doable and add support for some small oled screens.
Probably nicest solution.  It is a pretty low-overhead library (I had it running in systems with 32kbyte of combined instruction and data memory). I think we have enough space remaining for this.

I can implement this if needed.

However what we should display on the display is a 2nd questions we should answer first.

STemplar, 2023-01-22:
Indeed, I would say hardcode some variables first, like Relative Modulation level (17), Boiler flow water temperature (25) and Return water temperature (28). OEM Diagnostic code (115). Maybe later make it configurable via WebUI?

DaveDavenport, 2023-01-22:
https://user-images.githubusercontent.com/406245/213925010-9b9dfc65-883a-41fa-a43e-5178c29d524f.mp4

Roos-AID, 2023-01-22:
This is indeed nice add-on function, but for me it lower priority than extra sensors. How many gpio pins are extern available on the wemos ? We already need one for onewire temp sensors, one for s0 kwh counter and one is reserved for gpio output. We do need 2 more for the i2c.

DaveDavenport, 2023-01-22:
The i2c are already exposed and you can hang a multiple sensors + oled displays on this bus.

Roos-AID, 2023-01-23:
Dave, which IO pins are used for SDA and SCL on V2 and V2.8 boards ?  I can't find this in the manual

closed 2026-02-16 by rvdbreemen