rvdbreemen/OTGW-firmware #154: [Request] Suggestions to improve security

0crap, 2022-12-23:
To improve security of the OTGW, I have some suggestions that might help.

- Fix issue #131 
Go for a "simple" solution, like pressing the Wemos D1 reset button on powerup, create a initial flash which only allows AP mode once, ....
The Wemos D1 is in a socket for a reason. :-)
- Put a username and/or password on the GUI. (http://otgw.local)
- See if the GUI can do https. (A certificate error is fine, at least the data is still encrypted. Just the endpoint can't be verified.)
- Allow the user to disable telnet when it's not needed. (Stop the daemon or don't allow connecting to port 23.)
- See if Telnet can be improved to only allow a secure connection. (SSH, port 22.)
- Allow the user to disable OTmonitor when it's not needed. (Stop the daemon or don't allow connecting to port 25238.)

MQTT is a good example how it may look like.
[image]

In any case, we appreciate the effort and spare time that goes into projects like this.
Even if you can put in a few of the above, that would be great!

Thank you!

rvdbreemen, 2022-12-24:
Thanks for your suggestions. They are noted as enhancements. I will put them on my backlog for a to be release. 
If you could have one feature of the list you gave me, what would you implement?

0crap, 2022-12-24:
> Thanks for your suggestions. They are noted as enhancements. I will put them on my backlog for a to be release. If you could have one feature of the list you gave me, what would you implement?

The WiFi issue seems to be the biggest risk, but probably also the hardest to fix?
Next would be to disable unused daemons, or block incoming connections to those ports.
Close with that comes a username/password on the GUI, to prevent unauthorized access. (And keep the daemons on/off checkbox save.)
https and ssh probably last.

But really, just pick one that seems doable first. 🥇

DaveDavenport, 2022-12-24:
In my opinion the wifi creating a public AP  is the biggest issue as it is public.

The other items are limited to 'private' and should not be an issue for most. 
If an otgw is in a network where it could be an issue, this can be resolved  on local/network level. 
(In my network, otgw can access the mqtt server and the ntp server. I can access otgw and that is it).

- Adding ssh is most likely to much for the esp8266 to handle .

0crap, 2022-12-25:
> In my opinion the wifi creating a public AP is the biggest issue as it is public.
>
> The other items are limited to 'private' and should not be an issue for most. If an otgw is in a network where it [...]

Fair enough on the ssh part, scratch that from the list if not possible.
And I agree on the WiFi issue being the biggest one.

On the other hand I think the majority of the users has a "flat" LAN IP network.
It's good that you have restricted the communication possibilities of the OTGW on your LAN, but most users will never know how to do that.
Next to that, the OTGW being a LAN device should not mean we can totally ignore the security aspects of it.
It seems you looked at it very well, by restricting the traffic flow from/to it.

rotilho, 2022-12-28:
Thank you @rvdbreemen for all your hard work! Your product is fantastic. I wanted to also share my thoughts - in my opinion, the WiFi connection and unauthenticated access are the most pressing concerns.  If I had to prioritize them, I would start with addressing the WiFi issue. As for the GUI with HTTPS, using a reverse proxy could potentially be a potential solution. I don't have any strong opinions on the last two points. Thank you again for everything you do

0crap, 2023-01-14:
Related to #6 and #21

DaveDavenport, 2024-07-20:
```patch
diff --git a/networkStuff.h b/networkStuff.h
index c947e1f..d197c63 100644
--- a/networkStuff.h
+++ b/networkStuff.h
@@ -129,14 +129,16 @@ void startWiFi(const char* hostname, int timeOut)
   //--- if it does not connect it starts an access point with the specified name
   //--- here  "<HOSTNAME>-<MAC>"
   //--- and goes into a blocking loop awaiting configuration
-  OTGWSerial.printf("AutoConnect to: %s\r\n", thisAP.c_str());
-  if (!manageWiFi.autoConnect(thisAP.c_str()))
-  {
-    //-- fail to connect? Have you tried turning it off and on again? 
-    DebugTln(F("failed to connect and hit timeout"));
-    delay(2000);  // Enough time for messages to be sent.
-    ESP.restart();
-    delay(5000);  // Enough time to ensure we don't return.
+  if ( ! manageWiFi.getWiFiIsSaved()  || ESP.getResetReason() == "External System"){
+    OTGWSerial.printf("AutoConnect to: %s\r\n", thisAP.c_str());
+    if (!manageWiFi.startConfigPortal(thisAP.c_str()))
+    {
+      //-- fail to connect? Have you tried turning it off and on again?
+      DebugTln(F("failed to connect and hit timeout"));
+      delay(2000);  // Enough time for messages to be sent.
+      ESP.restart();
+      delay(5000);  // Enough time to ensure we don't return.
+    }
   }
 
   // WiFi.mode(WIFI_STA); // explicitly set mode, esp defaults to STA+AP
```

small patch that will only show wifi portal if no ap configured, or explicit reset  via reset button (for 4 minutes).

rvdbreemen, 2026-11-05:
Hi @0crap, thank you for the clear list, and thanks to @DaveDavenport and @rotilho for the input. This issue has been open a long time, so here is where each point stands for 2.0.0 (ESP32-S3), and what we will and will not do.

**1. Open WiFi setup portal when the router is offline (#131).** Partly done.
- A triple press of the reset button within 10 seconds already clears the WiFi credentials and opens the setup portal.
- From the next 2.0.0 beta on, the unit no longer opens the portal after its own WiFi watchdog restart; it keeps trying the saved network instead.
- What is left is a normal boot while the router is down: the unit then still opens the setup portal for a few minutes. We will add an opt-in setting so that a unit with saved WiFi never opens the portal and just keeps retrying (the idea from DaveDavenport's patch). The default stays as it is, because otherwise a unit that moves to a new router cannot be reconfigured without the triple reset or USB.

**2. Password on the web UI.** Done.
- Set an admin password under Settings (user name `admin`).
- It protects the UI, the settings, firmware and filesystem updates, reboot, WiFi reset and the settings file download. The API never sends a password back.
- The limits: the password is optional and off by default, it uses HTTP Basic Auth over plain HTTP, and read-only data (WebSocket, MQTT, OpenTherm values) stays open on purpose, because this is a trusted-LAN device.

**3. HTTPS.** Not planned. We decided against it: it costs RAM and CPU, a self-signed certificate still gives browser warnings, and the WebSocket and the integrations assume plain HTTP/WS. If you need encryption, put the gateway behind a reverse proxy or a VPN, and please keep it off the open internet.

**4. Disable telnet.** Yes, we will add a setting for port 23. Good point: the debug console has no login and accepts a few keys that change state, for example the output pin. It will stay on by default, so support captures keep working, and you can switch it off.

**5. Disable the OTmonitor port (25238).** Done, and off by default. Switch it on under Settings only when you need OTmonitor.

**6. SSH.** Agreed it is not realistic on this device, and you struck it yourself. Not planned.

Points 1 and 4 are on our backlog for 2.0.0. Thanks again for raising this. If you would like to test a build with those settings later, that would be very welcome.

rvdbreemen, 2026-11-11:
Both follow-ups from the plan above have shipped: a setting to switch the telnet debug console (port 23) on or off (2.0.0-beta.94), and an opt-in setting that skips the WiFi setup portal when the saved WiFi is only unreachable at boot (2.0.0-beta.96). Both are in the public 2.0.0 beta. Closing; thanks again for the suggestions.

closed 2026-11-11 by rvdbreemen

rotilho, 2026-11-11:
That's awesome, thank you so much @rvdbreemen! 🙌

Very appreciate you coming back to this issue after all these years and addressing the security concerns.

I know how much time and effort goes into maintaining an open source project, especially for free. Thanks again for all your hard work and dedication, OTGW is a fantastic project!