rvdbreemen/OTGW-firmware #157: OTGW MQTT message flooding

dtryba, 2023-01-04:
I noticed some issues with the mqtt autodiscovery plugin in domotiz since installation of the OTGW:
my zigbee devices became unreliable.

It turns out that with MQTT enabled on OTGW there is a flood of messages that domoticz just cannot handle in time (pi3b). The tcp connection to the mqtt server (not on the pi) has a constant backlog to process. Updates to the zigbee network can get delayed by up to 2 minutes.

My heater is a Daalderop Combifor 24/50+ and the thermostat is a Honeywell Home Chronotherm Touch Modulation. The amount of mqtt traffic is a constant 5kb/s to 10kb/s when enabled when the heater is idle.

I get the feeling that any and all message a propagated to MQTT even though there are no state changes (well the temperatures read can fluctuate at the rounding). In the past I did some processing of the PS output to dummy switches and mqtt (in 10s intervals).

My question: is there a state change mechanism in the code to avoid this kind of unnecessary message flood? Is it a "just me" problem due to hardware? Or would there be a need to filter the dupes (for some specific time)?

But here is a 5s sample of mqtt payloads, they don't show state changes so there would be no need to send the messages at all.

```
online
100.00
22.50
-D-O-W--
OFF
ON
OFF
ON
OFF
OFF
OFF
--------
OFF
OFF
OFF
OFF
OFF
OFF
OFF
OFF
-D-O-W--
OFF
ON
OFF
ON
OFF
OFF
OFF
--------
OFF
OFF
OFF
OFF
OFF
OFF
OFF
OFF
-D-O-W--
OFF
ON
OFF
ON
OFF
OFF
OFF
--------
OFF
OFF
OFF
OFF
OFF
OFF
OFF
OFF
-D-O-W--
OFF
ON
OFF
ON
OFF
OFF
OFF
--------
OFF
OFF
OFF
OFF
OFF
OFF
OFF
OFF
-D-O-W--
OFF
ON
OFF
ON
OFF
OFF
OFF
--------
OFF
OFF
OFF
OFF
OFF
OFF
OFF
OFF

```

rvdbreemen, 2023-01-05:
hi @dtryba 

_My question: is there a state change mechanism in the code to avoid this kind of unnecessary message flood? Is it a "just me" problem due to hardware? Or would there be a need to filter the dupes (for some specific time)?_

You have asked multiple questions:
1.  No, I have no state change mechanism. I did explore it a long time ago, but decided not to implement it, as the OT message come in 2x per second max. Which should be doable for any hardware IMHO.
2. I have not heard of anyone before that has processing issues you seem to experience.
3. You are welcome to implement the feature, there is a old dev branch that could be used as inspiration. ;-) 

Also there are other things to consider, for example, once only "update" on change is implemented. What do we do when Domoticz is updated, and the state will be rebuild. The design decision so far has been send out a message each time when there is OT message received.

Hope this helps understanding your situation.

Kind regards,
Robert

closed 2023-01-05 by rvdbreemen

rvdbreemen, 2023-01-05:
Re-reading your message once more. But you are saying that your Zigbee and your wifi are causing issues. It sure sounds to me like they are interfering. Have you verified that they are nicely separated channels?

I found this on the interference of Zigbee and wifi channels... https://www.metageek.com/training/resources/zigbee-wifi-coexistence/

You might wanna consider moving either wireless to another channel so they don't overlap.