rvdbreemen/OTGW-firmware #179: Add I2C display

rvdbreemen, 2023-01-29:
Display for basic information on the print using I2C display.
On display :  Date/Time, Uptime, IP address, last error? Boiler modulation?

DaveDavenport, 2023-01-29:
[image]
[image]

We need to decide what to display.
This can be tested on the branch I made. The code should be isolated enough so others can easily tweak it.

tjfsteele, 2023-01-29:
The PCB is designed to make adding a 4 pin i2c oled easy to do. Does need
software!

Tim

DaveDavenport, 2023-01-29:
Code is there now for a small 1.3" oled directly to the header.

tjfsteele, 2023-01-29:
Unfortunately some displays come with the pins in the correct order and
some don't!

Tim

DaveDavenport, 2023-01-29:
yes, that is annoying yes..  what is worse..  some have an SSD13xx controller, some an SH1106. They live on the same i2c adress , but behave *slightly* different.
So you get a distorted display if they don't match.

tjfsteele, 2023-01-29:
Is there a way to sense that in software?

DaveDavenport, 2023-01-29:
Not that I know off.  In general I found that the 0.96" have the ssd13xx drivers and the 1.3" have the sh1106 ones. 
These are compatible: https://www.tinytronics.nl/shop/en/displays/oled/1.3-inch-oled-display-128*64-pixels-white-i2c

DaveDavenport, 2023-01-29:
bar on right is for icons, showing flame/etc. But I need somebody to make the icon.

[image]

[image]
[image]
[image]

tjfsteele, 2023-01-29:
It would be useful to find out if the oled displays sold by the nodo shop
are compatible.

Robert has a good flame icon..

Tim

DaveDavenport, 2023-01-29:
hard to say from the page,  it looks like the 0.96" version.

DaveDavenport, 2023-01-29:
https://github.com/rvdbreemen/OTGW-firmware/pull/178/ if anybody wants to test.

DaveDavenport, 2023-01-29:
[image]
would probably look like this..

tjfsteele, 2023-01-29:
You could ask him to send one?

rvdbreemen, 2023-01-29:
The icon space is very small, a 'normal' icon is 16x16 or so.

DaveDavenport, 2023-01-29:
10x10 pixels should fit in there. it is 18 pixels wide.

rvdbreemen, 2023-01-29:
To create an icon for displays, do the following steps:
1. Open Gimp (install first: https://www.gimp.org/downloads/)
2. Find an icon on a site like (flaticon), download it, and open it in Gimp
3. Selecteer Image-->Mode-->Indexed-->Use black and white (1 bit) palette
4. Scale to correct size (16x16 of 32x32 etc.)
5. If the icon needs improvement, after scaling. Then touchup the icon.

Touch up icon:
6. Select the 'perceel brush'
7. Left to the brush (links), select the 1 pixel brush
8. Change the size to the 1 pixel
9. Touch up your icon to your liking 

Remove Alpha Transpancy Layer:
10. Remove transparency layer (Layer --> Transpancy --> Remove Alpha Layer)

Exporteer icon to XBM format
11. File --> Export as (CTRL-E)
12. Enter a new name, ending in '.xbm'
13. Open the result in notepad, then you can find:

<code example in xmb format>
#define flame_16x16_width 16
#define flame_16x16_height 16
static unsigned char flame_16x16_bits[] = {
   0x80, 0x00, 0x80, 0x00, 0x80, 0x01, 0xc0, 0x01, 0xc0, 0x01, 0xc0, 0x03,
   0xe0, 0x0b, 0xf0, 0x0f, 0x70, 0x1f, 0x78, 0x1e, 0x38, 0x1e, 0x38, 0x1c,
   0x18, 0x1c, 0x18, 0x1c, 0x30, 0x0e, 0x20, 0x07 };
<code>

Then goto this repo to see how to integrate the icons:
https://github.com/JDJelectronics/u8g2_Android_icons/blob/main/Statusbar_icons_android_U8G2/Statusbar_icons_android_U8G2.ino

Shout out to @JDJelectronics

DaveDavenport, 2023-01-29:
I know how to add icons, I just refuse to pick/draw icons as I always spend tons of time doing this, to then be told its wrong. So apparently I am not good at it.

> Find an icon on a site like (flaticon), download it, and open it in Gimp

rvdbreemen, 2023-01-29:
I will try to find some this week 😇 Don't worry about it.