rvdbreemen/OTGW-firmware #197: OTGW unresponsive

ixperiencenl, 2023-02-07:
Hi, thanks a lot for this firmware, it works most of the time with my Home Assistant setup.
I noticed however that the data sending through MQTT is having hickups many times a day. 

The items in Home Assistant then become unavailable and I cannot see anything sent from OTGW.

Just as of now, I can still ping the OTGW, but the telnet is unresponsive, and so is the web server. 
Any idea on how we can debug this and deliver you the info to help fix this issue?
I am running on FW 0.10.0+36108cf - which I compiled myself through the Arduino IDE (the esp-toolkit install failed).
Using ESP8266 on an older nodoshop gateway (I think the second version)

rvdbreemen, 2023-02-09:
The behavior you describe sounds like the MQTT connection is disconnected for some reason. 

The values in HA become unavailable when the device goes "offline" for some reason. Without more information I cannot diagnose what is going on. 

Possible reasons are: WiFi issues, causing disconnects, or instability of your mqtt server, older version of mqtt hub server, interference on WiFi, edge of network range. 

What you can do diagnose your issue:
- check to see if you see the device to offline and online (in home assistant)
- do some logging of the ESP firmware, just connect using Putty (or another WiFi client)
- look in the debug information webpage and share the content, one of the parameters there is "WiFi receive power" and share all the info on that page here

Share the information you gather here, that way I can give you more suggestions on what is going on. 
- look at boot count, to see if it reboots for some reason (normally you should see low reboot count)

The good news is that the firmware does recover in your situation, as it reconnects and starts sending information. That's what we like to see happen.

Let's debug and share your information here.

ixperiencenl, 2023-02-09:
It could have something to do with WiFi, but the web interface also timed out at those moments. Currently, it's working stable and no timeouts, so no idea what's really going on there. I do see there are quite some reboots. This is what the debug page spits out:

Developer Robert van den Breemen
Firmware Version 0.10.0+36108cf
PIC Available true
PIC Firmware Version 4.2.7
PIC Device ID pic16f88
PIC Firmware Type gateway
Compiled on (date/time)Feb 6 2023 20:05:17
HostName OTGW
IP address 192.168.
MAC address 
Free Heap Mem (bytes) 9680
Max. Free Mem (bytes) 8872
Unique Chip ID d055e1
Arduino Core Version 3.0.2
Espressif SDK Version 2.2.2-dev(38a443e)
CPU speed (MHz) 160
Sketch Size (bytes) 604448
Sketch Free (bytes) 1490944
Flash ID 001640D8
Flash Chip Size (MB)4
Real Flash Chip (MB)4
LittleFSsize 1
Flash Chip Speed (MHz) 40
Flash Mode DIO
Board Type Unknown board
Wifi Network (SSID)
Wifi Receive Power (dB)-85
NTP Enable true
NTP Timezone Europe/Amsterdam
Uptime Since Boot 1(d)-15:40(H:m)
Last Reset Reason External System
Nr. Reboots 15
MQTT Connected true
Thermostat Connected true
Boiler Connected true
Gateway/Standalone true
HA Integration true

rvdbreemen, 2023-02-09:
Your wifi reception is -85 dBm, which is very low. This will mean it can disconnect and be very unstable. Do you have an option to move your ESP closer to your accesspoint?

<html>
<body>

Signal Strength | TL;DR |   | Required for
-- | -- | -- | --
-30 dBm | Amazing | Max achievable signal strength. The client can only be a few feet from the AP to achieve this. Not typical or desirable in the real world. | N/A
-67 dBm | Very Good | Minimum signal strength for applications that require very reliable, timely delivery of data packets. | VoIP/VoWi-Fi, streaming video
-70 dBm | Okay | Minimum signal strength for reliable packet delivery. | Email, web
-80 dBm | Not Good | Minimum signal strength for basic connectivity. Packet delivery may be unreliable. | N/A
-90 dBm | Unusable | Approaching or drowning in the noise floor. Any functionality is highly unlikely. | N/A

</body>
</html>

closed 2024-01-25 by rvdbreemen