rvdbreemen/OTGW-firmware #206: Configured hostname is not used

mrtnbr, 2023-02-11:
Since updating to 0.10.0 the hostname I configure in settings is not actually used, it seems to be "esp-" with the last few digits of the MAC address added.

I have tried resetting all settings to default and then reconfiguring, rebooting, power cycling the ESP and OTGW, but nothing seems to make a difference. 

Config file has the right setting, I see the right setting in the UI but the actual hostname is different.

Some more info:
- NTP also does not seem to work
- Telnet works
- MQTT/HA works

[image]

[settings.ini.txt](https://github.com/rvdbreemen/OTGW-firmware/files/10713652/settings.ini.txt)
[debug.txt](https://github.com/rvdbreemen/OTGW-firmware/files/10713656/debug.txt)

rvdbreemen, 2023-02-11:
I assume that you also have rebooted the ESP? Or turned off and on? Just to make sure you tried that too.

mrtnbr, 2023-02-11:
Yes, I unplugged the power and plugged it back in after a minute.

rvdbreemen, 2023-02-11:
What’s you upgrade path so far. You went from 0.9.5 to 0.10.0 and now to 0.10.1? 

Also what’s your network setup, it kind of sounds like there is a network issue to me at this point. 

Could you share some logs from the esp?

mrtnbr, 2023-02-11:
Yes, 0.9.5 (and before that older), then 0.10.0, then 0.10.1

WiFi access point is a MikroTik cAP, router is a MikroTik RB5009. The ESP connects and shows up in the router, MQTT works well. Hostname just seems to not change.

I have flashed an unused spare ESP with the firmware, and it show the same behaviour. It might still be a network issue, or something in my workflow, but I guess that rules out any old stuff from previous versions.

Where can I find the ESP logs?

DaveDavenport, 2023-02-11:
I just tested this on mine.  Changed it from OTGW -> OTGW2.
Using avahi I see it on otgw2.local. So it should resolve fine under otgw2.local. But IDD my accesspoint reports esp-{numbers}.

NTP works fine for me, but I redirect all ntp traffic to my own ntp server.

Roos-AID, 2023-02-12:
Robert, I just flashed 0.10.1+56994f1 and the hostname on the AP changed to ESP-xxxxxx 
The configured (OTGWELGA)  Hostname still resolves when http://otgwelga.local is used.  

It seems only to be related to the Wifi AP connection.

rvdbreemen, 2023-02-12:
As I have MikroTik here as well. I will go an check to see what my MikroTik sees. Not sure why this is happening now, as there has been no change in the last binary. So it must be introduced when going to 0.10.x

mrtnbr, 2023-02-12:
> As I have MikroTik here as well. I will go an check to see what my MikroTik sees. Not sure why this is happening now, as there has been no change in the last binary. So it must be introduced when going to 0.10.x

Thanks. Yes, for me 0.9.5 worked as expected, when I upgraded to 0.10.0 it changed. 0.10.1 did not make a difference from 0.10.0

rvdbreemen, 2023-02-12:
@mrtnbr Thanks for reporting this. I have to look into this myself, there is only once change that I can think off to be honest. In the mean time I can confirm myself too that I see this issue.

We will figure this out.

mrtnbr, 2023-02-12:
> @mrtnbr Thanks for reporting this. I have to look into this myself, there is only once change that I can think off to be honest. In the mean time I can confirm myself too that I see this issue.
> 
> We will figure this out.

Thank you! I have updated the opening post, my NTP not working was caused by a misconfiguration on my DHCP server, it was giving out a non-working NTP server.

Roos-AID, 2023-02-12:
Robert, I think it has to do with Wifi library version.  I have my test unit still running on 0.10.0  compiled by myself and this one does not show the behaviour.  Unfortunatly I migrated my VSC environment to latest versions so I cannot check which Wifi lib version I used.

rvdbreemen, 2023-02-12:
Well, it seems that it is indeed the wifi library issue. But also the order of things, because I moved the NTPstart() back after the Wifi Start... and it now works again. So I have a fix, but we lose overwritting the NTP setting using DHCP again.

brhahlen, 2023-02-13:
I think this also affects me.
I use authentication for MQTT, but since the upgrade to 0.10.0, I got an "client not authorized" error, and it showed "OTWG<MAC_ADDRESS>" as the requesting client, instead of OTGW, which I had defined.

rvdbreemen, 2023-02-14:
This issue has been fixed. The next release will resolve the issue. If you like to help test / validate this. Please come to the discord and ask to be added to the beta channel (if your not already).

mrtnbr, 2023-02-14:
> This issue has been fixed. The next release will resolve the issue. If you like to help test / validate this. Please come to the discord and ask to be added to the beta channel (if your not already).

I have tested the new build and I can confirm the problem is fixed for me. Thank you!

rvdbreemen, 2023-02-15:
Thanks @mrtnbr I will go and do a release soon, with just this single fix.

closed 2023-02-15 by rvdbreemen