rvdbreemen/OTGW-firmware #208: Support thermostats that ignore remote override.

mcc05, 2023-02-20:
Wondering if it would be possible in this firmware to add support so that thermostats that ignore msg9 remote override and get the firmware to step in as many seem not to support it.

My thoughts:
A flag on the firmware config that one could set to indicate if the connected thermostat supports remote override or not. If so normal operation continues.

If not when the firmware get one of the remote override instructions (permanent/temporary) it takes over and issues the commands to the gateway.
i.e. stores the new setpoint, monitors the thermostat actual temp for on off, makes a prediction of the control set point, I suppose acts like a basic thermostat :)

DaveDavenport, 2023-02-20:
There are projects (on OTGW discord server these are discussed/developed) that implement a thermostat via OTGW.
Trying to 'tweak' output of a thermostat is probably a bad idea as it would throw off the control algorithm.

mcc05, 2023-02-20:
Hi Dave, I'm suggesting the OTGW firmware takes over the role as the thermostat in cases where one want to override a connected one that doesn't support it. Even basic functions like on/off would be good for driving things like geolocation from HA etc.

hvxl, 2023-02-20:
You should understand that a thermostat is not a simple device that just tells the boiler the desired room temperature. It contains a sophisticated algorithm that calculates the water temperature necessary to achieve the desired room temperature, within a reasonable time, without too much overshoot, in an energy efficient way.

I suppose it is possible to implement such an algorithm in an ESP8266 and only use the thermostat as a room temperature sensor and display. But that will take quite some resources. That sounds more like an alternative project, rather than something to try and squeeze into this project.

mcc05, 2023-02-20:
I do, but could just start with a hysteresis algo, CS set to flow temp setting like in a non modulating setup, not that efficient I know, or get more fancy into PWM/TPI, or heating curves/profiles like weather comp stored in setting files?
Yep ESP8266 might be an issue, but ESP32 might open up possibilities.

rvdbreemen, 2023-02-22:
Great idea. It has simply not been the focus of this project, this has started as a simple replacement is ESPlink and improvement of the ‘native’ integrations I saw in home assistant and Domoticz home automation projects. 

I personally have no plans to add this. But I welcome pull requests. I would suggest you go onto the Discord and join the SAT channel as some of our community have build an advanced thermostat in home assistant already. You could learn from them what it takes to do this. 

Some thoughts on your request:
* you can run this project and the OTGW hardware in standalone mode, meaning just don’t connect a thermostat and it will continue to function, then you can control the boiler using the commands of OTGW
* be warned this will take a significant part of your spare time, so just know what you are getting yourself into 😉
* the is plenty of room on the esp8266 I think to build this feature, before putting in the effort to move to esp32, just go an try a prototype (let’s not forget you only ever need 640kB a wise man said 🤔)

Enjoy your journey, please do a PR when you have results to share. I would love to see a SAT based on the esp8266. 

I just won’t go there,

Robert

closed 2023-02-22 by rvdbreemen

hvxl, 2023-02-22:
>     * you can run this project and the OTGW hardware in standalone mode, meaning just don’t connect a thermostat and it will continue to function, then you can control the boiler using the commands of OTGW
> 
There is no need to disconnect the thermostat. You can control the boiler just as well with a thermostat connected. I would actually advise to keep it connected. First of all, it can provide you with the room temperature as input for your control algorithm, assuming the thermostat reports it. Secondly, it can function as a fail safe mechanism if your code stops functioning (or you hit the bail out button when it goes haywire).

>     * the is plenty of room on the esp8266 I think to build this feature, before putting in the effort to move to esp32, just go an try a prototype (let’s not forget you only ever need 640kB a wise man said)
> 
I think you're mistaken there. Allegedly it was Bill Gates who said that.

DaveDavenport, 2023-02-22:
if only we had a massif amount of 640kb ram as that 'wise man' once said was enough.
From the esp8266 doc:

```
According to our current version of SDK, SRAM space available to users is assigned as
below.
• RAM size < 50 kB, that is, when ESP8266EX is working under the Station mode and
connects to the router, the maximum programmable space accessible in Heap +
Data section is around 50 kB.
```

rvdbreemen, 2023-02-23:
Hmmm, now I remember why I wanted to challenge myself on this particular platform. So far it's been a amazing journey to build this on the esp platform. Also moving to an esp32 would give space and room for more feature. So yeah, Port it an do a Pull Request or fork it all together 😇