rvdbreemen/OTGW-firmware #210: Wiki - How to use it .... Homey now supported too!

gruijter, 2023-03-06:
https://github.com/rvdbreemen/OTGW-firmware/wiki/How-to-use-it-with-Home-Assistant-or-Domoticz-or-others

Could you add Homey as a supported platform? I posted instructions on how to use with Homey here:
https://community.homey.app/t/77190

Thx for your work!

rvdbreemen, 2023-03-07:
@gruijter thank you for adding support. I have added it to the wiki documentation, thanks for building the support for Homey and adding value to this eco-system around OpenTherm Gateway.

Robert

closed 2023-03-07 by rvdbreemen

gruijter, 2023-03-08:
Thx!
The newly created link o the WiKi is dead though...

rvdbreemen, 2023-03-08:
That's because the page changed: 

https://github.com/rvdbreemen/OTGW-firmware/wiki/How-to-use-it-with-Home-Assistant,-Homey-or-Domoticz-or-others

You should be proud I think.

rvdbreemen, 2023-03-09:
Got it, fixed.

rvdbreemen, 2023-03-09:
And I will fix the NL version later ;-)