rvdbreemen/OTGW-firmware #211: Flashing ino & fs on nodemcu (2018) makes web interface unavailable

jtebbens, 2023-03-06:
Hi,

I used esplink wifi firmware for a long time and wanted to step in to MQTT type of data communication to test with Homey OTGW MQTT. I tried flashing several versions and when I flash and keep it on the Microusb cable powered it works. I get wifi, I can access the webinterfce etc. But when I then move the wifi module to the OTGW board and powered it with 24v adapter I can ping the gateway but the web interface doesnt load.

jtebbens, 2023-03-06:
Developer
Robert van den Breemen
ESP8266 Firmware Version
0.9.5+4eb7d24 (30-05-2022)
OTGW PIC Firmware Version
OTGW PIC Device ID
Compiled on (date/time)
May 30 2022 08:35:36
HostName
OTGW
IP address
192.168.1.19
MAC address
CC:50:E3:08:E2:5A
Free Heap Memory (bytes)
13064
Max. Free Memory (bytes)
11792
Unique Chip ID
8e25a
Arduino Core Version
2_7_4
Espressif SDK Version
2.2.2-dev(38a443e)
CPU speed (MHz)
160
Sketch Size (bytes)
553.906
Free Sketch Space (bytes)
1492
Flash ID
00164068
Flash Chip Size (MB)
4
Real Flash Chip Size (MB)
4
LittleFSsize
2
Flash Chip Speed (MHz)
40
Flash Mode
DIO
Board Type
ESP8266_NODEMCU
Wifi Network (SSID)
stardust
Wifi Receive Power (dB)
-58
NTP Enable
true
NTP Timezone
Europe/Amsterdam
Uptime Since Boot
0(d)-00:00(H:m)
Last Reset Reason
External System
Nr. Reboots
2
MQTT Connected
true
Thermostat Connected
false
Boiler Connected
false
Gateway mode
false
OTGW Online/Offline
true

rvdbreemen, 2023-03-07:
@jtebbens in your report you say that you tried several version. The current flashed version seems to be an old version 0.9.5. I recommend to use the latest version, as the 0.10.x branch supports all pic variants. 

You can find that here: https://github.com/rvdbreemen/OTGW-firmware/releases/tag/v0.10.1

When you remove the esp from the board, you say it does work. Could you go and check to see what is in the reboot.log on the ESP and share that here.

Sometimes it works better erase the flash completely, and reinstall from that point. Also please make sure to flash both FS and INO binaries, to make sure you have a matching filesystem and binary firmware installed.

Do you have any indication on the wifi quality once installed in the OTGW board? 

Hope to hear from you, 
Robert

jtebbens, 2023-03-07:
I tried 0.10.x too but the board fails on the OTGW socket. I can only ping it but very unstable as well.

rebootlog.txt

2023-03-07 12:28:20 - reboot cause: External System (6) 
External Reason: External Watchdog reason: 
2023-03-07 12:24:19 - reboot cause: External System (6) 
External Reason: External Watchdog reason: 

Developer
Robert van den Breemen
Firmware Version
0.10.1+56994f1
PIC Available
false
PIC Firmware Version
no pic found
PIC Device ID
no pic found
PIC Firmware Type
no pic found
Compiled on (date/time)
Feb 11 2023 09:16:10
HostName
OTGW
IP address
192.168.1.19
MAC address
CC:50:E3:08:E2:5A
Free Heap Mem (bytes)
11312
Max. Free Mem (bytes)
9568
Unique Chip ID
8e25a
Arduino Core Version
3.0.2
Espressif SDK Version
2.2.2-dev(38a443e)
CPU speed (MHz)
160
Sketch Size (bytes)
601104
Sketch Free (bytes)
1495040
Flash ID
00164068
Flash Chip Size (MB)
4
Real Flash Chip (MB)
4
LittleFSsize
1
Flash Chip Speed (MHz)
40
Flash Mode
DIO
Wifi Network (SSID)
stardust
Wifi RX Power (dBm)
-56
Wifi Quality (%)
95
Wifi Quality
Amazing
NTP Enable
true
NTP Timezone
Europe/Amsterdam
Uptime Since Boot
0(d)-00:00(H:m)
Last Reset Reason
External System
Nr. Reboots
1
MQTT Connected
false
Thermostat Connected
false
Boiler Connected
false
Gateway/Standalone
false
HA Integration
true

jtebbens, 2023-03-07:
Erased flash:
esptool.exe -cp COM4 -cd nodemcu -ce

Install ino & fs bin files again. AP configured back to correct wifi. Works with microusb but still fails on socket board 24v.

2023-03-07 13:49:30 - reboot cause: External System (6) 
External Reason: External Watchdog reason: 
2023-03-07 13:46:19 - reboot cause: External System (6) 
External Reason: External Watchdog reason:

rvdbreemen, 2023-03-07:
The rebootlog.txt shows that the system is reset by the external watchdog. 

If you would join the Discord community, we could maybe have more minds join in on the issue solving.
To join discord, please go here: https://discord.gg/ggkzcpfc7d

jtebbens, 2023-03-07:
Joined

closed 2024-01-25 by rvdbreemen