rvdbreemen/OTGW-firmware #212: Cant set Max CH setpoint

richardscholten73, 2023-03-10:
I am unable to change the Max CH setpoint.

My boiler is set to a maximum temperature of 80°C.
OTGW has a (default?) Max CH setpoint of 70°C

I tried setting it to match my boilers temp of max temp of  80°C with otmonitor in the Heater configuration menu and with a Free format command (SH=80).
I also tried setting it with MQTT explorer. (MaxTSet).

Nothing works, the Max CH setpoint just stays at 70°C

When updating the Control Setpoint in Home assistant with SERVICE OPENTHERM_GW.SET_CONTROL_SETPOINT OTGW reports the correct Control Setpoint, however my boiler receives a Control Setpoint of the default 70°

```
ESP8266 Firmware Version0.9.5+4eb7d24 (30-05-2022)
OTGW PIC Firmware Version6.2
OTGW PIC Device IDpic16f1847
Compiled on (date/time)May 30 2022 08:35:36
HostNameOTGW
IP address <removed>
MAC address <removed>
Free Heap Memory (bytes)12344
Max. Free Memory (bytes)11512
Unique Chip ID5aca9c
Arduino Core Version2_7_4
Espressif SDK Version2.2.2-dev(38a443e)
CPU speed (MHz)160
Sketch Size (bytes)553.906
Free Sketch Space (bytes)1492
Flash ID0016405E
Flash Chip Size (MB)4
Real Flash Chip Size (MB)4
LittleFSsize2
Flash Chip Speed (MHz)40
Flash ModeDIO
Board TypeESP8266_NODEMCU
Wifi Network (SSID)Wi-Fi1
Wifi Receive Power (dB)-79
NTP Enabletrue
NTP TimezoneEurope/Amsterdam
Uptime Since Boot0(d)-04:15(H:m)
Last Reset ReasonExternal System
Nr. Reboots18
MQTT Connectedtrue
Thermostat Connectedfalse
Boiler Connectedtrue
Gateway modetrue
OTGW Online/Offlinetrue
```

hvxl, 2023-03-10:
OTGW has a default Max CH setpoint of 90°C. If it doesn't accept the SH=80 command (response SH: BV), it must be because the boiler has informed it of other limits via MsgID 49. Possibly that information is old, but is not being polled by the thermostat. You can try to force an update with a PM=49 command.

richardscholten73, 2023-03-10:
The PM=49 reports back:
18:29:32.169190	B4031501E	Read-Ack  	Max CH setpoint boundaries: 80 30

But the Max CH watersetpoint remain 70: 
18:29:39.148867	B40394600	Read-Ack  	Max CH water setpoint: 70.00

hvxl, 2023-03-10:
The SH command is now accepted by the OTGW (I don't know if it wasn't before, your report wasn't clear on that). The Read-Ack that shows 70 degrees is from an earlier time stamp than the SH command. So that doesn't give any indication whether the command worked or not. The OTGW should send a Write-Data for the Max CH water setpoint. But it will only do that if the boiler indicated the Max CH water setpoint can be written, via ID6:LB1.

richardscholten73, 2023-03-10:
I dont see a write command in the log after sending the SH command.

"The OTGW should send a Write-Data for the Max CH water setpoint. But it will only do that if the boiler indicated the Max CH water setpoint can be written, via ID6:LB1."
Not sure if my boiler (intergas Hreco 36) supports this.
But thats not the issue here i think? 
I am not trying to set the Max settemp of my boiler through OTGW.
My boiler is set to 80 and thats fine, its OTGW that is stuck on 70 for some reason.

You mentioned the default is 90, but somehow it is now set to 70.
Where is the 70 comming from?
Is there any way to go back to the default setting?
Complete reset of the device/factory reset?

richardscholten73, 2023-03-10:
So the B indicates the value actualy comes from my boiler, I am so confused right now...

I set all values i can set on my boiler back to the defaults as described in the boiler manual.
I disconnected/reconnected both OTGW and my boiler from power.  (just to make sure there are no values left in memory)

PM=49 now returns Max CH setpoint boundarie 90 30, as expected as these are the defaults.
But the max setpoint is returned as 70.
None of the setting on my boiler even had or have 70 as a temp...

```
22:26:48.056884	Command: PM=49
22:26:48.225090	PM: 49
22:26:48.572219	R90013BDD	Write-Data	Control setpoint: 59.86
22:26:48.764315	B50013BDD	Write-Ack 	Control setpoint: 59.86
22:26:49.569421	R80380000	Read-Data 	DHW setpoint: 0.00
22:26:49.760089	B40383C00	Read-Ack  	DHW setpoint: 60.00
22:26:50.569387	R00390000	Read-Data 	Max CH water setpoint: 0.00
22:26:50.755982	B40394600	Read-Ack  	Max CH water setpoint: 70.00
22:26:51.568330	R80310000	Read-Data 	Max CH setpoint boundaries: 0 0
22:26:51.756208	B40315A1E	Read-Ack  	Max CH setpoint boundaries: 90 30
```

BTW, i do not have a thermostat attached to OTGW. Everything is controlled through the Homeassistant integration and appDeamon.
Only command that is send to OTGW with the integration is a call to OPENTHERM_GW.SET_CONTROL_SETPOINT service with the Control setpoint temp (59.86 in the log above)
Could the homeassitant integration be causing this?

richardscholten73, 2023-03-10:
I found it.

Sifted through the whole manual of my boiler and it was a setting on my boiler after all.
A different setting in a different menu...
Why this setting is in a different menu, and why there are two seperate settings for the Max setpoint is beyond me.

Thank you for your assistants and patiens.
Sorry for wasting your time.

Kind regards,
Richard Scholten

closed 2023-03-10 by richardscholten73

hvxl, 2023-03-10:
This cannot be caused by the homeassitant integration. It clearly is a setting on the boiler.

Reading the installation manual of the Kombi Kompakt HReco, I understand things as follows:
1. Parameter 5. sets the upper limit for the Max CH setpoint (default 90)
2. Parameter E sets the lower limit for the Max CH setpoint (default 40)
3. You set the actual Max CH setpoint via the control panel: Press the &#127777; button until the LED next to the radiator symbol lights up. Then you can set the temperature between 30°C and 90°C (standard setting 80°C).