rvdbreemen/OTGW-firmware #215: Settings menu not loading

jcouzy, 2023-04-07:
Hi,

I am a new user of the OTGW which I acquired from Node Shop.
I received it fully programmed including the WiFi module.
I was able to configure WiFi and am able to access the OTGW.
Everything seems to work fine, except the settings menu.
When I click on the settings button, I get the red bar with loading...... in it and it stays this way.

Any clue?
[image]

rvdbreemen, 2023-04-07:
Did you also flash the filesystem?

jcouzy, 2023-04-10:
Yes, also reflashed the file system, to be sure there are no custom configuration active which could interfere.

rvdbreemen, 2023-04-10:
So can you use the file explorer (right navigation menu) to check to see if you can find the setting.ini file.

rvdbreemen, 2023-04-15:
How did it go @jcouzy ?

jcouzy, 2023-04-19:
I was able to download, modify and upload the file, but it looks like the setting changes don't get activated.
I get the assumption that the WiFi module is very unstable.

rvdbreemen, 2024-01-25:
@jcouzy did you ever look at the debug menu, it tells you the WiFi quality.

closed 2024-01-25 by rvdbreemen