rvdbreemen/OTGW-firmware #217: Home Assistant 20223.5.0 now throwing exception

mcc05, 2023-05-04:
As of the latest version of HA, the status string returned from OTG is throwing an exception

Logger: homeassistant.components.mqtt.models
Source: components/sensor/__init__.py:583
Integration: MQTT ([documentation](https://www.home-assistant.io/integrations/mqtt), [issues](https://github.com/home-assistant/home-assistant/issues?q=is%3Aissue+is%3Aopen+label%3A%22integration%3A+mqtt%22))
First occurred: 06:42:09 (2631 occurrences)
Last logged: 07:40:29

Exception raised when updating state of sensor.idealboiler_status_slave, topic: 'IdealBoiler/value/idealC40/status_slave' with payload: b'-C-F----'
Exception raised when updating state of sensor.idealboiler_status_slave, topic: 'IdealBoiler/value/idealC40' with payload: b'online'
Exception raised when updating state of sensor.idealboiler_status_master, topic: 'IdealBoiler/value/idealC40' with payload: b'online'
Exception raised when updating state of sensor.idealboiler_status_master, topic: 'IdealBoiler/value/idealC40/status_master' with payload: b'-D---W--'
Exception raised when updating state of sensor.idealboiler_status_slave, topic: 'IdealBoiler/value/idealC40/status_slave' with payload: b'--------'
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/components/sensor/__init__.py", line 579, in state
    numerical_value = int(value)
ValueError: invalid literal for int() with base 10: '-D---W--'

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/components/mqtt/models.py", line 270, in process_write_state_requests
    entity.async_write_ha_state()
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 585, in async_write_ha_state
    self._async_write_ha_state()
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 649, in _async_write_ha_state
    state = self._stringify_state(available)
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 591, in _stringify_state
    if (state := self.state) is None:
  File "/usr/src/homeassistant/homeassistant/components/sensor/__init__.py", line 583, in state
    raise ValueError(
ValueError: Sensor sensor.idealboiler_status_master has device class None, state class None unit  and suggested precision None thus indicating it has a numeric value; however, it has the non-numeric value: -D---W-- (<class 'str'>)

mcc05, 2023-05-04:
Done a change for merge to stop this

brhahlen, 2023-05-04:
Can confirm and can also confirm that the changes made in #218 (at least on the lines 136 and 137) stop these errors.

serrj-sv, 2023-05-08:
@rvdbreemen Could you please release binaries with fix

rvdbreemen, 2023-05-09:
Trying to fix CI pipeline... anyone any ideas what it fails, help would be welcome ;-)

@serrj-sv I want to fix the pipeline, but you can download mqttha.cfg from the data folder in the dev branch and upload it manually.

rvdbreemen, 2023-05-10:
Ben,
That makes a lot of sense, I missed that completely.
Thanks for the PR,
Robert

brhahlen, 2023-05-10:
> Ben,
> That makes a lot of sense, I missed that completely.
> Thanks for the PR,
> Robert

You're welcome!
Glad I have been able to give a little bit back :)

rvdbreemen, 2023-05-11:
I looked into the issue myself last night. It turns out that the 2023.5.0 is breaking not just my integration, a lot of integrations broke of this change in HA Core. I have seen this before, this will also mean that 2023.5.x will most likely fix this in the Core side, but it's good to clean up the mqttha.cfg integration a bit.

Please keep upgrading HA yourself too, it could also be that it will fix itself in 2-3 weeks time because they fix core.

Seriously we have seen this many times before with HA, move fast break often and fix. The HA core project is really good at this ;-)

euro2, 2023-05-31:
Not sure but this is in the logs. Didnt check logs for weeks:

`Logger: homeassistant.components.mqtt.models
Source: components/sensor/__init__.py:583
Integration: MQTT (documentation, issues)
First occurred: 10:40:21 (34 occurrences)
Last logged: 10:42:07

Exception raised when updating state of sensor.otgw_status_master, topic: 'OTGW/value/otgw-A4E57C95FAB1/status_master' with payload: b'-----W--'
Exception raised when updating state of sensor.otgw_status_slave, topic: 'OTGW/value/otgw-A4E57C95FAB1/status_slave' with payload: b'--------'
Exception raised when updating state of sensor.otgw_status_slave, topic: 'OTGW/value/otgw-A4E57C95FAB1' with payload: b'online'
Exception raised when updating state of sensor.otgw_status_master, topic: 'OTGW/value/otgw-A4E57C95FAB1' with payload: b'online'
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/components/sensor/__init__.py", line 579, in state
    numerical_value = int(value)
ValueError: invalid literal for int() with base 10: '-----W--'

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/components/mqtt/models.py", line 270, in process_write_state_requests
    entity.async_write_ha_state()
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 585, in async_write_ha_state
    self._async_write_ha_state()
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 649, in _async_write_ha_state
    state = self._stringify_state(available)
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 591, in _stringify_state
    if (state := self.state) is None:
  File "/usr/src/homeassistant/homeassistant/components/sensor/__init__.py", line 583, in state
    raise ValueError(
ValueError: Sensor sensor.otgw_status_master has device class None, state class None unit  and suggested precision None thus indicating it has a numeric value; however, it has the non-numeric value: -----W-- (<class 'str'>)
`

rik1984, 2023-06-11:
Same issue here, I'm running HA 2023.5.2 where it is not fixed yet. I don't understand the instructions referring to post #218 (https://github.com/rvdbreemen/OTGW-firmware/pull/218), Should I update the OTGW firmware, or manually change something in OTGW?

Sorry in case this is a beginner question, I just try to get rid of my HA error logs, and this one has count of 56269 occurences since 48 hours already

rvdbreemen, 2023-06-11:
@rik1984 the fix needs to be validated, but you can download the mqttha.cfg from the dev branch on github. 

You can find it right here: https://github.com/rvdbreemen/OTGW-firmware/blob/dev/data/mqttha.cfg
If you download it, you can manually to the ESP using the file explorer. 

I have already download the beta build here: 
https://github.com/rvdbreemen/OTGW-firmware/suites/12881434620/artifacts/695010907

If you have some time, then please test it for me. Worst thing that could happen is that it works, and then I can release it.

rik1984, 2023-06-12:
Thanks for your guidance. I have manually deleted mqttha.cfg from the file system and uploaded the new version. I can confirm that it solves the issue

tomm1ed, 2023-08-25:
> @rik1984 the fix needs to be validated, but you can download the mqttha.cfg from the dev branch on github.
>
> You can find it right here: [...]

Downloaded the mqttha.cfg and replaced it on LittleFS but issue persists on latest version of HA (Home Assistant 2023.8.3)

t0mas, 2023-09-20:
Tested with Home Assistant 2023.8.4 and it solves the error issue in the log. So far all sensors etc seem to work as normal.

borft, 2023-09-22:
Tested on HA 2023.9.2, works like a charm, otgw sensors look well and stable.

mikeg-de, 2023-09-28:
I still face the issue spamming logs like crazy. @rik1984 when deleting and upliading a new mqttha.cfg, did you changed anything in partiuclar?

I am using:

Home Assistant 2023.9.3
Supervisor 2023.09.2
Operating System 10.5
Frontend 20230911.0 - latest

`Logger: homeassistant.components.mqtt.models
Source: components/sensor/__init__.py:593
Integration: MQTT ([documentation](https://www.home-assistant.io/integrations/mqtt), [issues](https://github.com/home-assistant/core/issues?q=is%3Aissue+is%3Aopen+label%3A%22integration%3A+mqtt%22))
First occurred: 26 September 2023 at 10:52:28 (73848 occurrences)
Last logged: 13:27:38

Exception raised when updating state of sensor.otgw_status_slave, topic: 'OTGW/value/otgw-485519150FBB/status_slave' with payload: b'-C------'
Exception raised when updating state of sensor.otgw_status_master, topic: 'OTGW/value/otgw-485519150FBB/status_master' with payload: b'-D---W--'
Exception raised when updating state of sensor.otgw_status_slave, topic: 'OTGW/value/otgw-485519150FBB/status_slave' with payload: b'--------'
Exception raised when updating state of sensor.otgw_status_slave, topic: 'OTGW/value/otgw-485519150FBB' with payload: b'online'
Exception raised when updating state of sensor.otgw_status_master, topic: 'OTGW/value/otgw-485519150FBB' with payload: b'online'
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/components/sensor/__init__.py", line 586, in state
    numerical_value = int(value)
                      ^^^^^^^^^^
ValueError: invalid literal for int() with base 10: '-D---W--'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/components/sensor/__init__.py", line 589, in state
    numerical_value = float(value)
                      ^^^^^^^^^^^^
ValueError: could not convert string to float: '-D---W--'

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/components/mqtt/models.py", line 305, in process_write_state_requests
    entity.async_write_ha_state()
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 730, in async_write_ha_state
    self._async_write_ha_state()
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 830, in _async_write_ha_state
    state, attr = self._async_generate_attributes()
                  ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 771, in _async_generate_attributes
    state = self._stringify_state(available)
            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 736, in _stringify_state
    if (state := self.state) is None:
                 ^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/sensor/__init__.py", line 593, in state
    raise ValueError(
ValueError: Sensor sensor.otgw_status_master has device class 'None', state class 'None' unit '' and suggested precision 'None' thus indicating it has a numeric value; however, it has the non-numeric value: '-D---W--' (<class 'str'>)`

borft, 2023-09-28:
> > I still face the issue spamming logs like crazy. @rik1984 when deleting and upliading a new mqttha.cfg, did you changed anything in particular?
> 
> No I didn't do anything special, only following the instructions. I'm now running HA9.1 ans will without issues (not sure whether the mqttha.cfg file has been updated along with my HA updates or whether it is still in the version I changed it to, to be honest.

well, the mqttha.cfg file is on the otgw right? (more specifically on the wifi microcontroller on the otgw), so I don't see how that would be affected by updating ha ;)

rik1984, 2023-09-28:
> well, the mqttha.cfg file is on the otgw right? (more specifically on the wifi microcontroller on the otgw), so I don't see how that would be affected by updating ha ;)

You are right indeed ;) so no changes done there since

rvdbreemen, 2023-12-10:
I will have a look into this soon. To see how to fix this.  It is most likely a fix to the mqttha.cfg templates that is needed.

rvdbreemen, 2024-01-25:
There is a working mqttha.cfg in the devbranch and there are dev binaries candidate available. 

Issue does not break HA integration. 

But I do need to release this. 🙈

closed 2024-01-25 by rvdbreemen