rvdbreemen/OTGW-firmware #226: Web UI Gateway Mode always false

hotchkj, 2023-09-23:
I feel like I'm missing something quite obvious, but I've searched the documentation on both this site and the original OTGW, and nothing seems to add up. I've got a NodoShop OpenTherm gateway connected to boiler & thermostat, running MQTT and all seems generally reporting ok, I can see & set thermostat temperature etc. from Home Assistant.

However, whenever I attempt to enable Gateway Mode, nothing immediately appears to change in the Web UI. I've tried via the MQTT set command & directly using OTMonitor. Both show a GW=1 command received. Yet the web UI continues to report `Gateway/Standalone false`. According to OTMonitor, the device does appear to be in Gateway Mode, but it's hard to be confident of what is reporting what status:

```
11:45:00.760867	Command: GW=1
11:45:00.818253	GW: P
11:45:00.823888	Command: PR=M
11:45:00.835738	T10D50000	Write-Data	Message ID 213: 0
11:45:00.880847	PR: M=G
```

Is this a bug in the web UI or is the device actually not operating as a gateway?

Advice appreciated; I may just be misunderstanding something. This remains quite a technical/complex topic!

hvxl, 2023-09-23:
Reading the source code, it seems this flag indicates whether an A or R message was seen in the last 30 seconds. Such messages may not occur, even in gateway mode, if the list of alternatives is empty or when the thermostat doesn't request any message IDs the boiler doesn't support.

rvdbreemen, 2024-01-25:
@hotchkj I think I can improve the documentation here. 

Like @hvxl correctly explained the flag is actually not based on the status, by querying the gateway, but by monitoring the msgids. 

This method is not completely reliable, and should be seen as indication only.  

Needs improving in a future release.

closed 2026-01-28 by rvdbreemen