rvdbreemen/OTGW-firmware #227: Adding commands to serial data causes problems with native HA integration

mvn23, 2023-10-18:
This was brought to my attention in https://github.com/home-assistant/core/issues/101518.
By sending commands (e.g. `SC`) from the ESP firmware to the OpenTherm Gateway directly, some unexpected data is received by Home Assistant. As a result we're seeing warnings in the logs and commands from Home Assistant seeming to time out (although they are probably working fine).

As this is the first option described on the [wiki](https://github.com/rvdbreemen/OTGW-firmware/wiki/How-to-use-it-with-Home-Assistant,-Homey-or-Domoticz-or-others#serial-2-network), it is likely that most users will try to connect via the serial connection first.

While I am planning to increase compatibility with different setups and multiple connections on the Home Assistant/pyotgw side, I'm still unsure on how to make it robust (is that `SE` for our command or for someone else's?). I appreciate that adding an option to disable any influence on the protocol may be quite a bit of work, but for now a note on your wiki would be nice ;)

rvdbreemen, 2024-01-25:
@mvn23 

Sorry for the late response. I can look into maximized legacy mode, surprising the time commands. 

Reading thru the tread on HA issue does not seem to be the only issue. The user had NTP traffic blocked. 

Thanks for letting me know, 
Robert

rvdbreemen, 2026-01-28:
This issue is solved in release 1.0.0 with the introduction of compatibility mode.