rvdbreemen/OTGW-firmware #228: OTGW OTA firmware upgrade failed

infoedkok, 2023-10-20:
Hoi,
With OTMONITOR, trying to upgrade firmware from 4.2.5. to 4.3.

Connecting to the gateway is succesfull, I see values in the monitor

It doesn't work: it showes message "firmware upgrade failed- 0 errors".
No entry in the logfile.
[image]

What could be the reason?

Regards, Ed

rvdbreemen, 2023-10-21:
Hi Ed

Just login into the webUI and use the OTGW firmware on the ESP to upgrade the PIC in the OTGW. 

That will work. Is there a reason you want to just upgrade to 4.3? As there is 5.x out for you hardware. 

Hope this helps,
Robert

rvdbreemen, 2024-01-25:
Closed due to inactivity.

closed 2024-01-25 by rvdbreemen