rvdbreemen/OTGW-firmware #229: OTGW HW reboot cause: Exception (2) - Access to invalid address (29)

TomEkk33, 2023-10-22:
Hello,

I have OTGW from NodoShop and Wifii WeMos D1 Mini with its firmware dated 20-03-2023 (from NodoShop)

I see some issues , one of them is related to FW reboot.

In reboot_log.txt of "FSexplorer ESP" I found:
```
2023-10-16 14:49:28 - reboot cause: Exception (2) - Access to invalid address (29)
ESP register contents: epc1=0x4000df64, epc2=0x00000000, epc3=0x00000000, excvaddr=0x00000000, depc=0x00000000
2023-10-12 12:14:58 - reboot cause: Exception (2) - Access to invalid address (29)
ESP register contents: epc1=0x4000df64, epc2=0x00000000, epc3=0x00000000, excvaddr=0x00000000, depc=0x00000000
2023-10-10 05:07:34 - reboot cause: Hardware Watchdog (1) 
ESP register contents: epc1=0x40103f2a, epc2=0x00000000, epc3=0x00000000, excvaddr=0x00000000, depc=0x00000000
2023-10-10 00:40:56 - reboot cause: Hardware Watchdog (1) 
ESP register contents: epc1=0x40104281, epc2=0x00000000, epc3=0x00000000, excvaddr=0x00000000, depc=0x00000000
2023-10-07 21:55:59 - reboot cause: Exception (2) - Access to invalid address (29)
ESP register contents: epc1=0x4000df64, epc2=0x00000000, epc3=0x00000000, excvaddr=0x00000000, depc=0x00000000
2023-10-07 13:31:16 - reboot cause: Exception (2) - Access to invalid address (29)
ESP register contents: epc1=0x4000df64, epc2=0x00000000, epc3=0x00000000, excvaddr=0x00000000, depc=0x00000000
```
Would you mind to check whether is it significant issue?

More details is described here: https://github.com/home-assistant/core/issues/101518

rvdbreemen, 2024-01-25:
Hi @TomEkk33 

I have looked at this issue. But I have no explanation on what you see. 

The crash log is not useful. Also read the thread on HA issue. 

I would not advise both using the serial integration and the MQTT at the same time. 

The fact that the firmware does sent some time commands should not interfere. If this however is confirmed to be an issue, the I could consider a maximum compatibility mode for legacy support. 

Hope to hear from you if your issue is resolved as you seem also to have an unsupported thermostat. 

Thanks 
Robert

TomEkk33, 2024-01-29:
Thank you for your answer. This issue appeared only once and never again. It was found just during the process of solving the
https://github.com/home-assistant/core/issues/101518. However the root cause of the issue has not been found.

closed 2024-01-29 by TomEkk33