rvdbreemen/OTGW-firmware #230: Ping problems

Nifrabar, 2023-11-01:
In addition to issue #222
As I still had disconnect connect in Domoticz log file I removed the software and went back to the original software with just serial connection. From that moment all worked stable again as usual.
Next step was buying a new NodeMcu V2 and flashed again the new software. OK all seems to work. But after a while again the connection error was there. Now I started pinging and here strange effects happened.
1st Ping > good
2nd Ping >fault
3rd Ping >fault

It seems that this is causing the connection problems. Random good/fault Ping result.

Maybe that baudrate is too high?
As I cannot change any network setting from the application I got stuck new.
Are there other bin-files to try?
I am not able to compile thd source.
My Domoticz is very stable without the new OTGW software. I am having about 30 ESPs (easy)  and also about 15 ESP (tasmota) devices which are very stable. 

So what suggestions are there.

Thanks.
-Bart

rvdbreemen, 2023-11-01:
Bart,
I notice you are talking about the baud rate and network settings, but the firmware is designed for use with MQTT.  Even though there is a legacy support mode available, my firmware is not designed for serial use. 

If you have any issues, I recommend not using my firmware. You can opt for the EasyESP (used before) or the NodeMCU firmware by Bron Schelte. 

Connection errors and ping issues are most likely on another level in the ESP, which I cannot explain. The reconnected could indicate unstable wifi issues,.

Can you see the quality of your wifi settings (in de Debug Information) page? That gives a good indicator if is maybe an unstable WiFi issue.

Thanks,
Robert

Nifrabar, 2023-11-01:
Robert,
I will test with another esp next to the otgw.
Then I will do several ping-test alternating for both esp s and revert with my findings.
Would a serial connection not suffer from instable wifi?
regards,
Bart

rvdbreemen, 2024-01-25:
@Nifrabar 

I highly recommend the MQTT integration and not the legacy serial over WiFi/network integrations. 

My firmware was specifically designed to no longer having to deal with crappy WiFi. 

Did you solve this issue? Is so let me know. 

Kinda regards
Robert

pvossel, 2024-02-12:
I have 2 NodeMCU's. 
One with the latest OTGW FW and another for a [different project](https://github.com/eelcohn/nRF905-API). 

They are both near each other on my attic. Connected to the same IoT SSID. 
The other NodeMCU has a 100% successrate pinging. The OTGW one has a 50~ % successrate. 

I suspect it might have something to do with wifi protocol. That one is using 802.11G and the other one 802.11N.

DaveDavenport, 2024-02-12:
Just as a test, I did  a quick test on my OTGW and I get no ping loss. 

```
100 packets transmitted, 100 received, 0% packet loss, time 99162ms
rtt min/avg/max/mdev = 2.519/6.725/66.196/7.444 ms
```

I wonder why one sais G other N, given this is something the lower level firmware (from esp) deals with and not the otgw layer on top (to the best of my knowledge).

pvossel, 2024-02-12:
@DaveDavenport, 

I recently changed my firmware on my accesspoints (TP Link M4 Deco) to OpenWRT. 
OpenWRT is a bit more flexible in settings. 

Since OpenWRT it is having this problem. Therefore I suspect some 802.11 mode issue (on OpenWRT or NodeMCU side).

Nifrabar, 2024-02-12:
Any progress on improving ping-behaviour?
Should like to have the MQTT version but as it turned out for me to be unstable I keet using the serial OTGW connection.

DaveDavenport, 2024-02-12:
Its very hard to debug this. There are so many unknowns. 
* WiFi can be very finiky.  rotating or moving a module a small bit can make/break a stable connection. The environment can cause a lot of influence on this.
* Many bad clones of wemon/nodemcu boards exists and you might get a bad one even from reputable sellers. This is an absolute pain.
* esp8266 had compatibility issues with APs in the past, in my experience revision matters. 
* The wifi is all hidden behind a thick layer of abstractions and firmware..  There is not a lot of low levels things we can do, esp with the arduino layer in between.

@pvossel  would it be possible as a test to swap the nodemcu around? flash the other one with otgw and the otgw one with the other project?  This is very curious.

pvossel, 2024-02-19:
Small update, the problem for me totaly dissapeared. 
Ive made a TCPdump from my accesspoint to the ESP. I saw it had trouble with an ARP broadcast. 
I've changed multicast to unicast and am now I am half a day problem free. 

Will monitor it the coming days.

rvdbreemen, 2024-03-18:
Closing this as it seems to be a network issue that the ESP has problems with. Related to SDK most likely.

closed 2024-03-18 by rvdbreemen