rvdbreemen/OTGW-firmware #238: No MQTT updates after reboot HA or Domoticz

infoedkok, 2024-01-09:
Hi,
After rebooting my NAS with HA and Domoticz in docker containers, both HA and DOM do not accept the MQTT messages of the firmware anymore.

Only a hard restart of the firmware makes both HA and DOM to process the MQTT OTGW messages again.

Could you please add a reboot option in the firmware, so the OTGW can be rebooted / restarted via the webpage?

Keep up the could work and thank you a lot for the excellent piece of SW!

Regards, Ed

infoedkok, 2024-01-09:
Info off the firmware in use:
Firmware Version0.10.2+50c3ed2
PIC Availabletrue
PIC Firmware Version4.2.5
PIC Device IDpic16f88
PIC Firmware Typegateway
Compiled on (date/time)Mar 14 2023 23:02:12

rvdbreemen, 2024-01-09:
Hi Ed,

Thanks for the question. Yes, there is a simple way to reset the OTGW from the WebUI. However it's not on the main UI of the webapp, it's deeper down into the webUI.

If you hit the button on the top right called [Advanced], you will see dropdown menu. If you click the option [File system contents], then you will get to the file explorer screen. On the bottom of the that screen you will the option [ReBoot]. 

As a bonus, you can also reset your wifi settings for the ESP, in case you need that.

Hope this helps,
Robert

closed 2024-01-09 by rvdbreemen

rvdbreemen, 2024-01-10:
@infoedkok One more thing, it was pointed out on the Discord community, that turning OFF the option [MQTT Home Assistant reboot detection] could also work in this situation. 

It's in the [Settings] page, and looks like this:
[image]

rvdbreemen, 2024-01-10:
Let us know what worked in your situation.

infoedkok, 2024-01-24:
@rvdbreemen Thank you for explaining those options. All worked for me ;-)