rvdbreemen/OTGW-firmware #242: Status in API (v0/v1) seems always 0

jan-gerard, 2024-03-31:
It is already quite I while that I checked the operation of my smart quooker valve that prevents using DHW when someone is showering. It uses the API (http://otgw.fritz.box/api/v1/otgw/id/0) to retrieve the (2 byte) status, looking for the value of LSB bit 2 for the DHW operating state. 
I noticed this was no longer working, and then found that I always get a 0. The same happens for http://otgw.fritz.box/api/v0/otgw/ which also reports a json status. No information. Only 0. Meanwhile, http://otgw.fritz.box/api/v1/otgw/otmonitor reports {"name": "dhwmode", "value": "on", "unit": "", "epoch": 1711911744}, so the value is read correctly.
But it seems that the /id/ interface of /api/v1/ is not functioning as it should.
I cannot find where it goes wrong in the source code. I'm using version 0.10.2+50c3ed2.  (as reported by the OTGW web interface). 

Is there more information that could provide help troubleshooting this issue? Or do I do something wrong with the API? Other id's are working as expected, like temperature readings. Only the status (id=0) seems broken.

rvdbreemen, 2024-03-31:
Hi @jan-gerard 

Not sure what is going on in your situation. So can you share some of the debug logging, with debug option 2 (REST API) turned on?

Just goto port 23, then hit "2" (turn on API logging) and hit "3" (turn off MQTT), this way you can log whats going on.

Thanks,
Robert

jan-gerard, 2024-03-31:
Hi Robert,

Wow, that was a quick response! I made some debug logging. 

```
23:36:58.873434 (  16800| 10832) handleDebug (  66):
Debug RestAPI: true

```
DHW was off:
```
23:39:40.874482 (  16384| 10832) operator()  ( 108): in 'onNotFound()'!! [/api/v1/otgw/id/0] =>
23:39:40.876923 (  15040| 10832) operator()  ( 111): next: processAPI(/api/v1/otgw/id/0)
23:39:40.877762 (  15040| 10832) processAPI  (  31): from[192.168.1.139] URI[/api/v1/otgw/id/0] method[GET]
23:39:40.878767 (  15000| 10832) processAPI  (  44): >>word[0] => [], word[1] => [api], word[2] => [v1], word[3] => [otgw], word[4] => [id], word[5] => [0],
23:39:40.880065 (  15000| 10832) sendOTGWvalu( 152): Status = 0

23:39:44.603586 (  16360| 10832) operator()  ( 108): in 'onNotFound()'!! [/api/v1/otgw/otmonitor] =>
23:39:44.605874 (  15016| 10832) operator()  ( 111): next: processAPI(/api/v1/otgw/otmonitor)
23:39:44.606919 (  15016| 10832) processAPI  (  31): from[192.168.1.53] URI[/api/v1/otgw/otmonitor] method[GET]
23:39:44.607999 (  14976| 10832) processAPI  (  44): >>word[0] => [], word[1] => [api], word[2] => [v1], word[3] => [otgw], word[4] => [otmonitor],
23:39:44.609100 (  14976| 10832) sendOTmonito( 259): sending OT monitor values ...

23:39:49.106204 (  16416| 10832) operator()  ( 108): in 'onNotFound()'!! [/api/v0/otgw] =>
23:39:49.108498 (  15072| 10832) operator()  ( 111): next: processAPI(/api/v0/otgw)
23:39:49.109330 (  15072| 10832) processAPI  (  31): from[192.168.1.139] URI[/api/v0/otgw] method[GET]
23:39:49.110316 (  15048| 10832) processAPI  (  44): >>word[0] => [], word[1] => [api], word[2] => [v0], word[3] => [otgw],
23:39:49.111644 (  15048| 10832) sendOTGWvalu( 152): Status = 0
23:40:41.870772 (  16552| 10832) operator()  ( 108): in 'onNotFound()'!! [/api/v1/otgw/otmonitor] =>
23:40:41.873061 (  15208| 10832) operator()  ( 111): next: processAPI(/api/v1/otgw/otmonitor)
23:40:41.874098 (  15208| 10832) processAPI  (  31): from[192.168.1.139] URI[/api/v1/otgw/otmonitor] method[GET]
23:40:41.875187 (  15168| 10832) processAPI  (  44): >>word[0] => [], word[1] => [api], word[2] => [v1], word[3] => [otgw], word[4] => [otmonitor],
23:40:41.876280 (  15168| 10832) sendOTmonito( 259): sending OT monitor values ...
```

Then I switched on the hot water, and the web interface shows dhwmode is On:
```
23:41:54.301707 (  16488| 10832) operator()  ( 108): in 'onNotFound()'!! [/api/v1/otgw/otmonitor] =>
23:41:54.304029 (  15144| 10832) operator()  ( 111): next: processAPI(/api/v1/otgw/otmonitor)
23:41:54.305070 (  15144| 10832) processAPI  (  31): from[192.168.1.139] URI[/api/v1/otgw/otmonitor] method[GET]
23:41:54.306179 (  15104| 10832) processAPI  (  44): >>word[0] => [], word[1] => [api], word[2] => [v1], word[3] => [otgw], word[4] => [otmonitor],
23:41:54.307284 (  15104| 10832) sendOTmonito( 259): sending OT monitor values ...

23:41:58.161761 (  15816| 10832) operator()  ( 108): in 'onNotFound()'!! [/api/v0/otgw] =>
23:41:58.164881 (  15144| 10832) operator()  ( 111): next: processAPI(/api/v0/otgw)
23:41:58.165902 (  15144| 10832) processAPI  (  31): from[192.168.1.139] URI[/api/v0/otgw] method[GET]
23:41:58.166917 (  15120| 10832) processAPI  (  44): >>word[0] => [], word[1] => [api], word[2] => [v0], word[3] => [otgw],
23:41:58.168059 (  15120| 10832) sendOTGWvalu( 152): Status = 0

23:42:05.318491 (  16384| 10832) operator()  ( 108): in 'onNotFound()'!! [/api/v1/otgw/id/0] =>
23:42:05.320852 (  15040| 10832) operator()  ( 111): next: processAPI(/api/v1/otgw/id/0)
23:42:05.321712 (  15040| 10832) processAPI  (  31): from[192.168.1.139] URI[/api/v1/otgw/id/0] method[GET]
23:42:05.322920 (  15000| 10832) processAPI  (  44): >>word[0] => [], word[1] => [api], word[2] => [v1], word[3] => [otgw], word[4] => [id], word[5] => [0],
23:42:05.324295 (  15000| 10832) sendOTGWvalu( 152): Status = 0
```
But the Status value remains 0. Always. Also when the central heating is on, or flame status is on. 

Jan-Gerard

jan-gerard, 2024-03-31:
Btw, with debug option 1, I could check that the status bits in the OT messages are actually showing the correct values.

jan-gerard, 2024-03-31:
I just updated the firmware, but to no avail. Or would I need to upload a new littleFS image as well? Then my connection settings get flushed. I try to avoid that.

rvdbreemen, 2024-04-01:
Hmmmm, weird. 
Can you send some logs with option 1 turned on... You can share it using pastbin too.
And yes, you also need to update the filesystem every time. It changes from time to time to, like the index.js or the actual pic firmware.

rvdbreemen, 2024-04-01:
Try this beta build for me please:
https://github.com/rvdbreemen/OTGW-firmware/actions/runs/8505272296/artifacts/1373828822

Yes, flash both the firmware and the filesystem.

I think I found a solution for the reported issue. A single AND replaced by an OR. 

Let me know if that works for you.

jan-gerard, 2024-04-01:
Hi Robert, With the new firmware, it works flawless again. So that fixed the issue. I did not yet update the FS image, though.

jan-gerard, 2024-04-01:
Hi again. Now I installed the FS .bin file as well. However, my web interface is now broken. Fortunately, the web interface still works (WiFi credentials are still valid), but I cannot change settings.
The screen keeps refreshing, while it shows:
[image]
FSExplorer works, but the home screen and settings button don't show anything useful.
In the browser console, I see the error:
```
Uncaught ReferenceError: assignment to undeclared variable data
    refreshSettings http://otgw.fritz.box/index.js:462
    settingsPage http://otgw.fritz.box/index.js:146
    initMainPage http://otgw.fritz.box/index.js:76

```

rvdbreemen, 2024-04-02:
@jan-gerard You can roll back the filesystem to the last stable release. I'll look at the index.js and try and fix it.

rvdbreemen, 2024-04-02:
@jan-gerard okay, I rolled back index.js to last known working version. 

Try this beta, the filesystem is updated to flash:
https://github.com/rvdbreemen/OTGW-firmware/actions/runs/8526281679/artifacts/1378307830

Let me know if this works for you...

jan-gerard, 2024-04-02:
Excellent! Everything is working again as it should. I get the Status bytes with sensible values, and the web gui is working as normal again. Thanks!

rvdbreemen, 2024-04-02:
Thank you for confirming that. And this could just be a first release in a years time 🥇

closed 2024-04-02 by rvdbreemen