rvdbreemen/OTGW-firmware #246: Exception raised when updating state of sensor.otgw_status_maste

mikeg-de, 2024-05-05:
Hi,

I just updated HASSIO to the most recent version and noticed these exceptions

```
2024-05-05 08:34:36.897 ERROR (MainThread) [homeassistant.components.mqtt.models] Exception raised when updating state of sensor.otgw_status_master, topic: 'OTGW/value/otgw-485519150FBB/status_master' with payload: b'CD---W--'
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/components/sensor/__init__.py", line 658, in state
    numerical_value = int(value)
                      ^^^^^^^^^^
ValueError: invalid literal for int() with base 10: 'CD---W--'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/components/sensor/__init__.py", line 661, in state
    numerical_value = float(value)
                      ^^^^^^^^^^^^
ValueError: could not convert string to float: 'CD---W--'

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/components/mqtt/models.py", line 378, in process_write_state_requests
    entity.async_write_ha_state()
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 1009, in async_write_ha_state
    self._async_write_ha_state()
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 1132, in _async_write_ha_state
    state, attr, capabilities, shadowed_attr = self.__async_calculate_state()
                                               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 1067, in __async_calculate_state
    state = self._stringify_state(available)
            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 1015, in _stringify_state
    if (state := self.state) is None:
                 ^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/sensor/__init__.py", line 665, in state
    raise ValueError(
ValueError: Sensor sensor.otgw_status_master has device class 'None', state class 'None' unit '' and suggested precision 'None' thus indicating it has a numeric value; however, it has the non-numeric value: 'CD---W--' (<class 'str'>)
2024-05-05 08:34:42.795 ERROR (MainThread) [homeassistant.components.mqtt.models] Exception raised when updating state of sensor.otgw_status_slave, topic: 'OTGW/value/otgw-485519150FBB/status_slave' with payload: b'-C-F----'
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/components/sensor/__init__.py", line 658, in state
    numerical_value = int(value)
                      ^^^^^^^^^^
ValueError: invalid literal for int() with base 10: '-C-F----'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/components/sensor/__init__.py", line 661, in state
    numerical_value = float(value)
                      ^^^^^^^^^^^^
ValueError: could not convert string to float: '-C-F----'

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/components/mqtt/models.py", line 378, in process_write_state_requests
    entity.async_write_ha_state()
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 1009, in async_write_ha_state
    self._async_write_ha_state()
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 1132, in _async_write_ha_state
    state, attr, capabilities, shadowed_attr = self.__async_calculate_state()
                                               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 1067, in __async_calculate_state
    state = self._stringify_state(available)
            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 1015, in _stringify_state
    if (state := self.state) is None:
                 ^^^^^^^^^^
[... 124 lines cut ...]
Espressif SDK Version2.2.2-dev(38a443e)
CPU speed (MHz)160
Sketch Size (bytes)597840
Sketch Free (bytes)1499136
Flash ID0016405E
Flash Chip Size (MB)4
Real Flash Chip (MB)4
LittleFSsize1
Flash Chip Speed (MHz)40
Flash ModeDIO
Wifi Network (SSID)unsernest
Wifi RX Power (dBm)-55
Wifi Quality (%)96
Wifi QualityAmazing
NTP Enabletrue
NTP TimezoneEurope/Amsterdam
Uptime Since Boot9(d)-09:28(H:m)
Last Reset ReasonSoftware/System restart
Nr. Reboots2
MQTT Connectedtrue
Thermostat Connectedtrue
Boiler Connectedtrue
Gateway/Standalonefalse
HA Integrationtrue
```

rvdbreemen, 2024-05-05:
Did you update the file system when you updated to 0.10.3?

As this issue has already been fixed by an updated configuration file of mqtthd.cfg

rvdbreemen, 2024-05-05:
@mikeg-de I will close this issue, you can reopen it if this not solved the issue in your case.

closed 2024-05-05 by rvdbreemen

mikeg-de, 2024-05-05:
I thought I updated littlefs but maybe the upload of the mqttha.cfg cuased the issue. I just updated littlefs again an manually restored the settings. I will check the HA logs the coming week. If the issue persists, I will reopen the ticket. Thanks for your feedback @rvdbreemen 👍

rvdbreemen, 2024-05-05:
Let me know either way please, so I can be at peace that this issue is fixed for you ;-)

mikeg-de, 2024-05-05:
You can be in peace. I restarted everything and checked the logs. No trace of the issue.

rvdbreemen, 2024-05-05:
Phew, thanks for that report. Lets hope its stable for you from now on.