rvdbreemen/OTGW-firmware #254: AA command has no effect

chryss44, 2024-11-16:
Hello,
I use Boiler De-Dietrich MCX 24/28 BIC PLUS with NETATMO and OTGW  on Wemos with last nodo shop board.
I have make all parameters on OpenTherm Monitor and have MQTT server to monitor information.

All Firmwares are in last version :

- OTGW firmware version 0.10.3+e334c42
- diagnose.hex	version 2.1
- gateway.hex	version6.6
- interface.hex	version 2.0
- OpenTherm Monitor Version 6.6
- Gateway (board) V2.11
 
If i just plug my Boiler on OTW, everything is correct and i have all interesting information for me and when i plug Netatmo some informations disappears, normal (when i read notice on http://otgw.tclcode.com/firmware.html#cmdps).
Well, if i want to receive informations what i need, i use AA= command on OpenTherm Monitor and log is correct. 
But nothing appears on the OTWMonitor, and when i use DA= command to delete, some information return not error message, as if information not exist.

I push log with command AA and DA where for example DA=27 repeat 2 time have no error (strange) and DA=25 / DA=28 are a return NF for not found and is normal.
But no return with all this command :-( .

Thanks for your help

`22:09:00.736377	SC: 22:09/6
22:09:01.251074	T00000300	Read-Data 	Status (MsgID=0): 00000011 00000000
		   - CH enable: enabled (1)
		   - DHW enable: enabled (1)
		   - Cooling enable: disabled (0)
		   - OTC active: not active (0)
		   - CH2 enable: disabled (0)
		   - Summer/winter mode: winter (0)
		   - DHW blocking: unblocked (0)
22:09:01.398549	BC000030A	Read-Ack  	Status (MsgID=0): 00000011 00001010
		   - Fault indication: no fault (0)
		   - CH mode: active (1)
		   - DHW mode: not active (0)
		   - Flame status: flame on (1)
		   - Cooling status: not active (0)
		   - CH2 mode: not active (0)
		   - Diagnostic indication: no diagnostics (0)
		   - Electricity production: not active (0)
22:09:02.250346	T00000300	Read-Data 	Status (MsgID=0): 00000011 00000000
		   - CH enable: enabled (1)
		   - DHW enable: enabled (1)
		   - Cooling enable: disabled (0)
		   - OTC active: not active (0)
		   - CH2 enable: disabled (0)
		   - Summer/winter mode: winter (0)
		   - DHW blocking: unblocked (0)
22:09:02.402814	BC000030A	Read-Ack  	Status (MsgID=0): 00000011 00001010
		   - Fault indication: no fault (0)
		   - CH mode: active (1)
		   - DHW mode: not active (0)
		   - Flame status: flame on (1)
		   - Cooling status: not active (0)
		   - CH2 mode: not active (0)
		   - Diagnostic indication: no diagnostics (0)
		   - Electricity production: not active (0)
22:09:03.248335	T00000300	Read-Data 	Status (MsgID=0): 00000011 00000000
		   - CH enable: enabled (1)
		   - DHW enable: enabled (1)
		   - Cooling enable: disabled (0)
		   - OTC active: not active (0)
		   - CH2 enable: disabled (0)
		   - Summer/winter mode: winter (0)
		   - DHW blocking: unblocked (0)
22:09:03.402357	BC000030A	Read-Ack  	Status (MsgID=0): 00000011 00001010
		   - Fault indication: no fault (0)
		   - CH mode: active (1)
		   - DHW mode: not active (0)
		   - Flame status: flame on (1)
[... 1106 lines cut ...]
		   - Fault indication: no fault (0)
		   - CH mode: active (1)
		   - DHW mode: not active (0)
		   - Flame status: flame on (1)
		   - Cooling status: not active (0)
		   - CH2 mode: not active (0)
		   - Diagnostic indication: no diagnostics (0)
		   - Electricity production: not active (0)
22:11:44.241413	T00000300	Read-Data 	Status (MsgID=0): 00000011 00000000
		   - CH enable: enabled (1)
		   - DHW enable: enabled (1)
		   - Cooling enable: disabled (0)
		   - OTC active: not active (0)
		   - CH2 enable: disabled (0)
		   - Summer/winter mode: winter (0)
		   - DHW blocking: unblocked (0)
22:11:44.342773	BC000030A	Read-Ack  	Status (MsgID=0): 00000011 00001010
		   - Fault indication: no fault (0)
		   - CH mode: active (1)
		   - DHW mode: not active (0)
		   - Flame status: flame on (1)
		   - Cooling status: not active (0)
		   - CH2 mode: not active (0)
		   - Diagnostic indication: no diagnostics (0)
		   - Electricity production: not active (0)`

rvdbreemen, 2024-11-23:
Hi @chryss44 

Did you checkout:

http://otgw.tclcode.com/matrix.cgi#boilers

And

http://otgw.tclcode.com/matrix.cgi#thermostats

To validate if your boiler and thermostat actually support the messages you are asking for?

Let me know,
Robert

rvdbreemen, 2024-11-23:
@chryss44 

If your boiler does not support the message ID then pushing AA is not changing this. 

Also pushing the same AA command with an ID it will cause OTGW to store the same message ID, with no real effect. 

You also point to the PS command, in the PS=1 state there is a default set of parameters sent, that does not mean that your boiler/thermostat actually support the messages IDs. 

Also not sure if you already do, but integration with MQTT on Home Assistant will give you a complete set of parameters that your boiler and thermostat support fully automatically. 

Hope this helps you figuring out what is going on. 

Kind regards, 
Robert

hvxl, 2024-11-23:
Read about [how the OTGW uses available slots to request additional information](http://otgw.tclcode.com/firmware.html#additional). Your thermostat requests no information that the boiler does not support, so there are no available slots. You will have to free up some slots to allow the OTGW to send its own requests.

chryss44, 2024-11-23:
It's good, i declare unknown Id 49, 5 and 122 for place 25, 27 and 28.
It's not user friendly to find what i can be replace but i'm arrive.
Perfect, thanks a lot @hvxl .

chryss44, 2024-11-24:
t's good, i declare unknown Id 49, 5 and 122 for place 25, 27 and 28.

closed 2024-11-24 by chryss44