rvdbreemen/OTGW-firmware #255: Central_heating sensor activated by automatically heating domestic hot water

bjornmorsman, 2024-12-22:
Hello,

i have a strange situation with the software, when my boiler (itho base cube) is automatically heating dhw water the sensor central_heating gets activated. Buth the central heating is not running. When i turn on the hot water the sensor domestic_hot_water get active.

my boiler has multiple states for heating, heating hot water and heating cv.

Is this an bug in the software of the esp or is this normal behavior?

thanks in advance

rvdbreemen, 2024-12-22:
Hi @bjornmorsman 

It is not a bug, it is normal l that when you turn on hot water your hot water indicator is ON. The boiler is then heating. 

Hope that explains,
Robert

bjornmorsman, 2024-12-22:
Hi robert,

thats not what i ment, my central_heating indicator goes on when my dhw water is heating

thank you

rvdbreemen, 2024-12-22:
Hi @bjornmorsman 

Can you share some logging. So I can see the messages that are being sent?

Thanks
Robert

bjornmorsman, 2024-12-22:
Hi @rvdbreemen 

i added the log below. At 19:29:34 central_heating goes to on in home assistant and at 19:31:36 it goes to off. central_heating_enable and domestic_hot_water are off at that moment.

hope you can help.

````19:28:00.773888	SC: 19:28/7
19:28:01.110704	T80190000	Read-Data 	Boiler water temperature: 0.00
19:28:01.267444	BC0191500	Read-Ack  	Boiler water temperature: 21.00
19:28:02.109316	T10010A00	Write-Data	Control setpoint: 10.00
19:28:02.265529	BD0010A00	Write-Ack 	Control setpoint: 10.00
19:28:03.109274	T80000200	Read-Data 	Status: 00000010 00000000
19:28:03.251897	B40000200	Read-Ack  	Status: 00000010 00000000
19:28:04.109870	T00110000	Read-Data 	Relative modulation level: 0.00
19:28:04.266355	BC0110000	Read-Ack  	Relative modulation level: 0.00
19:28:05.615106	T900E6400	Write-Data	Maximum relative modulation level: 100.00
19:28:05.639793	B500E6400	Write-Ack 	Maximum relative modulation level: 100.00
19:28:06.102453	T80190000	Read-Data 	Boiler water temperature: 0.00
19:28:06.259816	BC0191500	Read-Ack  	Boiler water temperature: 21.00
19:28:07.093895	T10010A00	Write-Data	Control setpoint: 10.00
19:28:07.252772	BD0010A00	Write-Ack 	Control setpoint: 10.00
19:28:08.096838	T80000200	Read-Data 	Status: 00000010 00000000
19:28:08.250449	B40000200	Read-Ack  	Status: 00000010 00000000
19:28:09.092842	T00110000	Read-Data 	Relative modulation level: 0.00
19:28:09.236766	BC0110000	Read-Ack  	Relative modulation level: 0.00
19:28:10.614437	T10101400	Write-Data	Room setpoint: 20.00
19:28:10.639899	BD0101400	Write-Ack 	Room setpoint: 20.00
19:28:11.076096	T80190000	Read-Data 	Boiler water temperature: 0.00
19:28:11.232813	BC0191500	Read-Ack  	Boiler water temperature: 21.00
19:28:12.090141	T10010A00	Write-Data	Control setpoint: 10.00
19:28:12.241811	BD0010A00	Write-Ack 	Control setpoint: 10.00
19:28:13.074020	T80000200	Read-Data 	Status: 00000010 00000000
19:28:13.216174	B40000200	Read-Ack  	Status: 00000010 00000000
19:28:14.058540	T00110000	Read-Data 	Relative modulation level: 0.00
19:28:14.233366	BC0110000	Read-Ack  	Relative modulation level: 0.00
19:28:15.618673	T10181533	Write-Data	Room temperature: 21.20
19:28:15.643849	BD0181533	Write-Ack 	Room temperature: 21.20
19:28:16.066151	T80190000	Read-Data 	Boiler water temperature: 0.00
19:28:16.213932	BC0191500	Read-Ack  	Boiler water temperature: 21.00
19:28:17.051115	T10010A00	Write-Data	Control setpoint: 10.00
19:28:17.215791	BD0010A00	Write-Ack 	Control setpoint: 10.00
19:28:18.083851	T80000200	Read-Data 	Status: 00000010 00000000
19:28:18.239919	B40000200	Read-Ack  	Status: 00000010 00000000
19:28:19.049012	T00110000	Read-Data 	Relative modulation level: 0.00
19:28:19.200215	BC0110000	Read-Ack  	Relative modulation level: 0.00
19:28:20.604549	T00110000	Read-Data 	Relative modulation level: 0.00
19:28:20.634594	BC0110000	Read-Ack  	Relative modulation level: 0.00
19:28:21.046263	T80190000	Read-Data 	Boiler water temperature: 0.00
19:28:21.190550	BC0191500	Read-Ack  	Boiler water temperature: 21.00
19:28:22.028920	T10010A00	Write-Data	Control setpoint: 10.00
19:28:22.187472	BD0010A00	Write-Ack 	Control setpoint: 10.00
19:28:23.043000	T80000200	Read-Data 	Status: 00000010 00000000
19:28:23.186266	B40000200	Read-Ack  	Status: 00000010 00000000
19:28:24.025308	T00110000	Read-Data 	Relative modulation level: 0.00
19:28:24.185086	BC0110000	Read-Ack  	Relative modulation level: 0.00
19:28:25.599780	T00390000	Read-Data 	Max CH water setpoint: 0.00
19:28:25.627176	B40393200	Read-Ack  	Max CH water setpoint: 50.00
19:28:26.030625	T80190000	Read-Data 	Boiler water temperature: 0.00
19:28:26.184826	BC0191500	Read-Ack  	Boiler water temperature: 21.00
19:28:27.029496	T10010A00	Write-Data	Control setpoint: 10.00
19:28:27.176438	BD0010A00	Write-Ack 	Control setpoint: 10.00
19:28:28.013982	T80000200	Read-Data 	Status: 00000010 00000000
19:28:28.171892	B40000200	Read-Ack  	Status: 00000010 00000000
19:28:29.020054	T00110000	Read-Data 	Relative modulation level: 0.00
19:28:29.171438	BC0110000	Read-Ack  	Relative modulation level: 0.00
19:28:30.599499	T1002010D	Write-Data	Master configuration: 00000001 13
19:28:30.625867	BD002010D	Write-Ack 	Master configuration: 00000001 13
19:28:31.013986	T80190000	Read-Data 	Boiler water temperature: 0.00
19:28:31.153794	BC0191500	Read-Ack  	Boiler water temperature: 21.00
19:28:31.999305	T10010A00	Write-Data	Control setpoint: 10.00
[... 413 lines cut ...]
19:31:48.462510	B401A2400	Read-Ack  	DHW temperature: 36.00
19:31:48.488507	AC0090000	Read-Ack  	Remote override room setpoint: 0.00
19:31:49.305864	T80190000	Read-Data 	Boiler water temperature: 0.00
19:31:49.448018	BC0193680	Read-Ack  	Boiler water temperature: 54.50
19:31:50.589443	T10010A00	Write-Data	Control setpoint: 10.00
19:31:50.617101	BD0010A00	Write-Ack 	Control setpoint: 10.00
19:31:51.289321	T80000200	Read-Data 	Status: 00000010 00000000
19:31:51.450164	B40000200	Read-Ack  	Status: 00000010 00000000
19:31:52.289212	T00110000	Read-Data 	Relative modulation level: 0.00
19:31:52.451533	BC0110000	Read-Ack  	Relative modulation level: 0.00
19:31:53.292606	T900E6400	Write-Data	Maximum relative modulation level: 100.00
19:31:53.430759	B500E6400	Write-Ack 	Maximum relative modulation level: 100.00
19:31:54.274636	T80190000	Read-Data 	Boiler water temperature: 0.00
19:31:54.435133	B40193780	Read-Ack  	Boiler water temperature: 55.50
19:31:55.592803	T10010A00	Write-Data	Control setpoint: 10.00
19:31:55.619238	BD0010A00	Write-Ack 	Control setpoint: 10.00
[... cut ...] 00010 00000000
19:31:56.417583	B40000200	Read-Ack  	Status: 00000010 00000000
19:31:57.275190	T00110000	Read-Data 	Relative modulation level: 0.00
19:31:57.417967	BC0110000	Read-Ack  	Relative modulation level: 0.00
19:31:58.262461	T10101400	Write-Data	Room setpoint: 20.00
19:31:58.426963	BD0101400	Write-Ack 	Room setpoint: 20.00
19:31:59.250736	T80190000	Read-Data 	Boiler water temperature: 0.00
19:31:59.412503	B40193880	Read-Ack  	Boiler water temperature: 56.50
19:31:59.499183	SC: 19:32/7````

hvxl, 2025-01-08:
> At 19:29:34 central_heating goes to on in home assistant and at 19:31:36 it goes to off. central_heating_enable and domestic_hot_water are off at that moment.

That is exactly what the boiler claims, according to your log:
`19:29:33.949860	B4000020A	Read-Ack  	Status: 00000010 00001010`
So yes, CH mode (ID0:LB1) is on while CH enable (ID0:HB0) is off. That is unusual. But it is your boiler that reports this weirdness. I strongly believe the OTGW should present the information it is given and not silently "correct" data that does not align with the specs.

closed 2025-01-10 by rvdbreemen