rvdbreemen/OTGW-firmware #258: OTGW

rhtenhove, 2025-01-09:
A few days ago I received a fresh OTGW with a Wemos D1 (for WiFi) and everything prepared (soldered and flashed by nodo shop).

After connecting my devices (had to do this first after setting up wifi, as it would not boot without):
- Remeha Avanta Ace 28c
- Honeywell Round Modulation

And updating:
- gateway.hex from 6.5 to 6.6
- OTGW firmware from 0.10.2 to 0.10.3+e334c42 (incl. LittleFS, for mqtt discovery)

All seemed well, however in Home Assistant I get various warnings and errors (Command PR failed and timeouts) from the Opentherm Gateway integration. I gave it some time, and the log collected quite a lot of these.

At first I also had MQTT enabled, and was checking the web interface, but I understand combining these causes unreliable results. So I disabled MQTT, closed the web interface, power cycled the OTGW, and opened telnet to see what's going on.

The HA logbook is full of the devices switching to unknown and then back on again, and the OTGW thermostat climate entity will lose temperature and setpoint. Automations don't behave well because of all the status changes. In telnet I see several `Not processed, received from OTGW`  messages, but I don't know what these mean. I've added part of the log at the bottom of this issue.

> When I had both MQTT enabled and the web interface shown, the web interface would also randomly be lacking data, as well as MQTT showing null for (nearly) all sensors before it would start slowly repopulating it again, until it once again lost all data.

Given that there seemed little that could have gone wrong (it was mostly plug and play), I'm not sure where to look now for what's misbehaving. The OTGW has internet, dns, ntp, and access to the world if need be, no limitations there.

When it is working, it works just fine, data is read correctly, states are correctly changed (both thermostat and boiler). However randomly it will go bananas, and I can't seem to be able to relate this to anything.

Any ideas? 🙏

### Logging and debug

Home assistant logs:

```
Logger: pyotgw.commandprocessor  
Source: runner.py:154  
First occurred: January 7, 2025 at 19:05:55 (7988 occurrences)  
Last logged: 07:57:26

- Command PR failed with PS: 0, retrying...
- Unknown message in command queue: PR: A=OpenTherm Gateway 6.6
- Command PR failed with PR: A=OpenTherm Gateway 6.6, retrying...
- Unknown message in command queue: PR: P=Low power
- Command PR failed with PR: P=Low power, retrying...
```

and 

```
Logger: pyotgw.pyotgw  
Source: runner.py:154  
First occurred: January 7, 2025 at 19:05:55 (2322 occurrences)  
Last logged: 07:57:26

- Timed out waiting for command: PR, value: L.
- Timed out waiting for command: PR, value: G.
- Timed out waiting for command: PR, value: I.
- Timed out waiting for command: PR, value: S.
- Timed out waiting for command: PR, value: P.
```

Here's the debug output:

|                         |                         |
| ----------------------- | ----------------------- |
| Developer               | Robert van den Breemen  |
| Firmware Version        | 0.10.3+e334c42          |
| PIC Available           | true                    |
| PIC Firmware Version    | 6.6                     |
| PIC Device ID           | pic16f1847              |
| PIC Firmware Type       | gateway                 |
| Compiled on (date/time) | Apr 17 2024 05:03:34    |
| HostName                | OTGW                    |
| IP address              | 192.168.178.95          |
| MAC address             | -                       |
[... 114 lines cut ...]
08:49:59.505252 (  16312| 15568) handleOTGW  (1771): Net2Ser: Sending to OTGW: [PR=L] (4)  
08:49:59.507652 (  15048| 14272) handleOTGW  (1771): Net2Ser: Sending to OTGW: [PR=M] (4)  
08:49:59.524380 (  15800| 14272) checkOTGWcmd(1293): CmdQueue: Checking if command is in in queue [PR: L=FXOMPC] (12)  
08:49:59.535575 (  16472| 15568) checkOTGWcmd(1293): CmdQueue: Checking if command is in in queue [PR: M=G] (7)  
08:49:59.636106 (  16472| 15568) processOT   (1676): Thermostat        T00050000 (9)[MsgID=  5][READ_DATA       ] ASFflags = ASF flags[00000000] OEM faultcode [  0]  
08:50:00.410694 (  16472| 15568) doTaskMinute( 279): Minute changed:  
08:50:00.413650 (  15128| 14272) sendOTGW    (1343): Sending to Serial [SC=8:50/4] (9)  
08:50:00.439784 (  15128| 14272) checkOTGWcmd(1293): CmdQueue: Checking if command is in in queue [SC: 08:50/4] (11)  
08:50:00.673543 (  16472| 15568) processOT   (1676): Boiler            BC00500FF (9)[MsgID=  5][READ_ACK        ]>ASFflags = ASF flags[00000000] OEM faultcode [255]  
08:50:00.688682 (  16472| 15568) processOT   (1676): Thermostat        T80190000 (9)[MsgID= 25][READ_DATA       ] Tboiler = 0.00 °C  
08:50:00.755065 (  17736| 16216) handleOTGW  (1771): Net2Ser: Sending to OTGW: [PR=M] (4)  
08:50:00.767580 (  15048| 14272) handleOTGW  (1771): Net2Ser: Sending to OTGW: [PS=1] (4)  
08:50:00.772786 (  15800| 14272) checkOTGWcmd(1293): CmdQueue: Checking if command is in in queue [PR: M=G] (7)  
08:50:00.823648 (  16472| 14272) checkOTGWcmd(1293): CmdQueue: Checking if command is in in queue [PS: 1] (5)  
08:50:00.141049 (  15128| 14272) processOT   (1707): Not processed, received from OTGW => (00000011/00000010,44.90,00000011/00000011,0.00,0.00,100.00,0/0,19.00,0.00,0.00,0.00,  
0.00,17.41,48.00,0.00,0.00,0.00,0.00,0,65/35,80/25,0.00,80.00,00000000/00000000,0,0,28024,31999,8846,17461,716,1996,15871,198) [209]  
08:50:00.144220 (  15048| 14272) handleOTGW  (1771): Net2Ser: Sending to OTGW: [PS=1] (4)  
08:50:00.156322 (  15128| 14272) checkOTGWcmd(1293): CmdQueue: Checking if command is in in queue [PS: 1] (5)  
08:50:00.185475 (  16392| 15568) handleOTGW  (1771): Net2Ser: Sending to OTGW: [PR=Q] (4)  
08:50:00.266271 (  16472| 15568) processOT   (1707): Not processed, received from OTGW => (00000011/00000010,44.90,00000011/00000011,0.00,0.00,100.00,0/0,19.00,0.00,0.00,0.00,  
0.00,17.41,PR: Q=E) [102]  
08:50:00.274975 (  17064| 15568) handleOTGW  (1771): Net2Ser: Sending to OTGW: [PR=Q] (4)  
08:50:00.357357 (  16472| 15568) processOT   (1707): Not processed, received from OTGW => (48.00,0.00,0.00,0.00,0.00,0,65/35,80/25,0.00,80.00,00000000/00000000,0,0,28024,PR: Q  
=E) [86]
```

rvdbreemen, 2025-01-10:
Hi @rhtenhove
Thanks for reporting. The logs you share show that you are using the legacy integration, using the serial over network connection. The legacy support in my firmware is just that, it supports the old serial over network option of using HA integration. However I build the firmware initially to make it more reliable, so using MQTT and auto discovery.

So turn off the Native HA integration of the OTGW component, thus stop using the serial over network integration. And then just use the MQTT option. Using both serial and MQTT at the same time seems to give you problems, so try stop using the "serial integration" to find out if it is more reliable.

Let me know what you find,
Robert

rhtenhove, 2025-01-11:
Just after posting this issue I had removed the legacy OTGW integration, and moved everything to MQTT. It functions well when it does, however my logs look like this. The most prominent one is that at 05:59:27 today it became unavailable and stayed as such until I power cycled the device.

| unavailable| gaps in timeline |
|-|-|
|[image]|[image]|

The access point mentioned the device was still connected, and all of the leds on the OTGW blinked as before. The opentherm thermostat also still worked fine to control the central heating. However only the power cycle brought it back online.

What provides a clue is that my MQTT broker (HASS Mosquitto Add-on v6.4.1) shows this message when it disconnects:

```
2025-01-08 21:05:40: Client OTGW disconnected due to malformed packet.
```

However the one this morning with the long disconnect shows

```
Client OTGW has exceeded timeout, disconnecting.
```

Doing another telnet log, I see a not processed from OTGW and right after MQTT breaks.

```
10:37:36.672646 (  15592| 15192) processOT   (1676): Request Boiler    R807A0000 (9)[MsgID=122][READ_DATA       ] DHWPumpValveOperationHours = 0 hrs
10:37:42.049264 (  15488| 15192) processOT   (1676): Boiler            BC07A3E24 (9)[MsgID=122][READ_ACK        ]>DHWPumpValveOperationHours = 15908 hrs
10:37:47.647437 (  15592| 15192) processOT   (1676): Answer Thermostat AC0090000 (9)[MsgID=  9][READ_ACK        ]>TrOverride = 0.00 °C
10:37:47.677440 (  15592| 15192) processOT   (1676): Thermostat        T00390000 (9)[MsgID= 57][READ_DATA       ] MaxTSet = 0.00 °C
10:37:52.822070 (  15592| 15192) processOT   (1676): Boiler            BC0395000 (9)[MsgID= 57][READ_ACK        ]>MaxTSet = 80.00 °C
10:37:53.927115 (  15592| 15192) processOT   (1707): Not processed, received from OTGW => (00) [2]
10:37:53.930609 (  14248| 13896) processOT   (1676): Thermostat        T80190000 (9)[MsgID= 25][READ_DATA       ] Tboiler = 0.00 °C
10:37:54.487514 (  16264| 15192) processOT   (1676): Boiler            B407B00C7 (9)[MsgID=123][READ_ACK        ]>DHWBurnerOperationHours = 199 hrs
10:37:54.583802 (  17800| 15192) processOT   (1707): Not processed, received from OTGW => (44CC) [4]
10:37:54.587773 (  16456| 15192) sendMQTTData( 406): Error: MQTT broker not connected.
10:37:54.589060 (  16456| 15192) processOT   (1676): Answer Thermostat AC0090000 (9)[MsgID=  9][READ_ACK        ]>TrOverride = 0.00 °C
10:37:54.600911 (  17800| 15192) sendMQTTData( 406): Error: MQTT broker not connected.
10:37:54.602865 (  16456| 15192) processOT   (1676): Thermostat        T10015000 (9)[MsgID=  1][WRITE_DATA      ]>TSet = 80.00 °C
10:37:54.612072 (  17800| 15192) processOT   (1676): Boiler            BD0015000 (9)[MsgID=  1][WRITE_ACK       ] TSet = 80.00 °C
10:37:54.622337 (  17800| 15192) sendMQTTData( 406): Error: MQTT broker not connected.
10:37:54.624386 (  16456| 15192) sendMQTTData( 406): Error: MQTT broker not connected.
10:37:54.625108 (  16456| 15192) sendMQTTData( 406): Error: MQTT broker not connected.
10:37:54.625662 (  16456| 15192) sendMQTTData( 406): Error: MQTT broker not connected.
10:37:54.626648 (  16456| 15192) sendMQTTData( 406): Error: MQTT broker not connected.
```

And it appears to recover when `CmqQueue` does its thing:

```
10:38:35.311634 (  15784| 15192) processOT   (1676): Boiler            BC000030C (9)[MsgID=  0][READ_ACK        ]>Status = Slave  [--WF----]
10:38:35.436739 (  17800| 15192) processOT   (1676): Thermostat        T00110000 (9)[MsgID= 17][READ_DATA       ] RelModLevel = 0.00 %
10:38:36.298575 (  17800| 15192) sendMQTTData( 406): Error: MQTT broker not connected.
10:38:36.300705 (  16456| 15192) processOT   (1676): Boiler            BC0112100 (9)[MsgID= 17][READ_ACK        ]>RelModLevel = 33.00 %
10:38:36.465624 (  17800| 15192) processOT   (1676): Thermostat        T00090000 (9)[MsgID=  9][READ_DATA       ] TrOverride = 0.00 °C
10:38:36.477559 (  17800| 15192) processOT   (1676): Request Boiler    R00740000 (9)[MsgID=116][READ_DATA       ] BurnerStarts = 0 
10:38:36.492029 (  17800| 15192) sendMQTTData( 406): Error: MQTT broker not connected.
10:38:36.494215 (  16456| 15192) processOT   (1676): Boiler            B40746E2B (9)[MsgID=116][READ_ACK        ]>BurnerStarts = 28203 
MQTT connected
10:38:37.149432 (  17432| 15192) addOTWGcmdto(1214): CmdQueue: Adding cmd end of queue, slot [0]
10:38:37.151396 (  16088| 15192) addOTWGcmdto(1217): CmdQueue: Insert queue in slot[0]:cmd[CS=unavailable] (14)
10:38:37.153803 (  16088| 15192) addOTWGcmdto(1237): CmdQueue: Next free queue slot: [1]
10:38:37.163930 (  17608| 15192) addOTWGcmdto(1214): CmdQueue: Adding cmd end of queue, slot [1]
10:38:37.166284 (  16264| 15192) addOTWGcmdto(1217): CmdQueue: Insert queue in slot[1]:cmd[CS=0] (4)
10:38:37.167516 (  16264| 15192) addOTWGcmdto(1237): CmdQueue: Next free queue slot: [2]
10:38:37.294600 (  16936| 15192) processOT   (1676): Answer Thermostat AC0090000 (9)[MsgID=  9][READ_ACK        ]>TrOverride = 0.00 °C
10:38:37.476775 (  16936| 15192) processOT   (1676): Thermostat        T00030000 (9)[MsgID=  3][READ_DATA       ] SlaveConfigMemberIDcode = Slave Config[00000000] MemberID code [  0]
10:38:38.290325 (  15592| 15192) processOT   (1676): Boiler            B4003410B (9)[MsgID=  3][READ_ACK        ]>SlaveConfigMemberIDcode = Slave Config[01000001] MemberID code [ 11]
10:38:38.511945 (  16936| 15192) processOT   (1676): Thermostat        T80190000 (9)[MsgID= 25][READ_DATA       ] Tboiler = 0.00 °C
[... cut ...]
10:41:45.843351 (  17800| 15192) processOT   (1676): Boiler            BD002010D (9)[MsgID=  2][WRITE_ACK       ] MasterConfigMemberIDcode = Master Config[00000001] MemberID code [ 13]
10:41:46.031892 (  17800| 15192) processOT   (1676): Thermostat        T80190000 (9)[MsgID= 25][READ_DATA       ] Tboiler = 0.00 °C
10:41:46.841797 (  17800| 15192) sendMQTTData( 406): Error: MQTT broker not connected.
```

But again CmdQueue saves the day a little later.

However still not sure what's going wrong, other than that somehow it sends an out of spec message.

rhtenhove, 2025-01-15:
To further the investigation, I've done a packet capture, and was "lucky" to receive a malformed packet and a timeout quickly after each other. The problem still persists, and it's quite frustrating since it makes it a lot harder to manage my multiple rooms and TRVs.

The malformed packet shows 3 packets which happened at the same time the Mosquitto logs show the failure. The first 2 appear fine, while the third seems to be unrecovered packet loss? Based on the `TCP Previous segment not captured`.

The timeout log was shown 25 seconds after this packet was seen on the line. Same thing, `TCP Previous segment not captured`, however that warning is shown numerous times throughout the full capture, where not all show a retransmission.

This was captured on a Raspberry Pi 5 in the Mosquitto Add-on container of Home Assistant OS using:

```shell
tcpdump -i eth0 -w - "(tcp port 1883) and (tcp[24:4]=0x4f544757 and tcp[28:4]=0x2f76616c and tcp[32:4]=0x75652f6f and tcp[36:4  
]=0x7467772d and tcp[40:4]=0x39344239 and tcp[44:4]=0x37453135 and tcp[48:4]=0x43413833)"
```

The latter is to filter on the topic `OTGW/value/otgw-94B97E15CA83` so it only captures packets destined for that topic.

<details>

<summary>Malformed packet</summary>

```ruby
No.     Time           Source                Destination           Protocol Length Info
   1528 1259.951098    192.168.178.95        172.30.33.0           MQTT     102    Publish Message [OTGW/value/otgw-94B97E15CA83/RelModLevel]

Frame 1528: 102 bytes on wire (816 bits), 102 bytes captured (816 bits) on interface -, id 0
    Section number: 1
    Interface id: 0 (-)
        Interface name: -
        Interface description: Standard input
    Encapsulation type: Ethernet (1)
    Arrival Time: Jan 15, 2025 13:12:53.489809000 CET
    UTC Arrival Time: Jan 15, 2025 12:12:53.489809000 UTC
    Epoch Arrival Time: 1736943173.489809000
    [Time shift for this packet: 0.000000000 seconds]
    [Time delta from previous captured frame: 11.522867000 seconds]
    [Time delta from previous displayed frame: 11.522867000 seconds]
    [Time since reference or first frame: 1259.951098000 seconds]
    Frame Number: 1528
    Frame Length: 102 bytes (816 bits)
    Capture Length: 102 bytes (816 bits)
    [Frame is marked: True]
    [Frame is ignored: False]
    [Protocols in frame: eth:ethertype:ip:tcp:mqtt]
    [Coloring Rule Name: TCP]
    [Coloring Rule String: tcp]
Ethernet II, Src: 02:42:60:0b:dc:f6 (02:42:60:0b:dc:f6), Dst: 02:42:ac:1e:21:00 (02:42:ac:1e:21:00)
    Destination: 02:42:ac:1e:21:00 (02:42:ac:1e:21:00)
        .... ..1. .... .... .... .... = LG bit: Locally administered address (this is NOT the factory default)
        .... ...0 .... .... .... .... = IG bit: Individual address (unicast)
    Source: 02:42:60:0b:dc:f6 (02:42:60:0b:dc:f6)
        .... ..1. .... .... .... .... = LG bit: Locally administered address (this is NOT the factory default)
        .... ...0 .... .... .... .... = IG bit: Individual address (unicast)
    Type: IPv4 (0x0800)
    [Stream index: 0]
Internet Protocol Version 4, Src: 192.168.178.95, Dst: 172.30.33.0
    0100 .... = Version: 4
    .... 0101 = Header Length: 20 bytes (5)
    Differentiated Services Field: 0x00 (DSCP: CS0, ECN: Not-ECT)
        0000 00.. = Differentiated Services Codepoint: Default (0)
        .... ..00 = Explicit Congestion Notification: Not ECN-Capable Transport (0)
    Total Length: 88
    Identification: 0x2a64 (10852)
    000. .... = Flags: 0x0
        0... .... = Reserved bit: Not set
        .0.. .... = Don't fragment: Not set
        ..0. .... = More fragments: Not set
    ...0 0000 0000 0000 = Fragment Offset: 0
    Time to Live: 254
    Protocol: TCP (6)
    Header Checksum: 0x5215 [validation disabled]
    [Header checksum status: Unverified]
[... 780 lines cut ...]
00e0  33 2f 6f 74 63 5f 61 63 74 69 76 65 4f 46 46 30   3/otc_activeOFF0
00f0  2c 00 27 4f 54 47 57 2f 76 61 6c 75 65 2f 6f 74   ,.'OTGW/value/ot
0100  67 77 2d 39 34 42 39 37 45 31 35 43 41 38 33 2f   gw-94B97E15CA83/
0110  63 68 32 5f 65 6e 61 62 6c 65 4f 46 46 30 32 00   ch2_enableOFF02.
0120  2d 4f 54 47 57 2f 76 61 6c 75 65 2f 6f 74 67 77   -OTGW/value/otgw
0130  2d 39 34 42 39 37 45 31 35 43 41 38 33 2f 73 75   -94B97E15CA83/su
0140  6d 6d 65 72 77 69 6e 74 65 72 74 69 6d 65 4f 46   mmerwintertimeOF
0150  46 30 2e 00 29 4f 54 47 57 2f 76 61 6c 75 65 2f   F0..)OTGW/value/
0160  6f 74 67 77 2d 39 34 42 39 37 45 31 35 43 41 38   otgw-94B97E15CA8
0170  33 2f 64 68 77 5f 62 6c 6f 63 6b 69 6e 67 4f 46   3/dhw_blockingOF
0180  46 30 33 00 29 4f 54 47 57 2f 76 61 6c 75 65 2f   F03.)OTGW/value/
0190  6f 74 67 77 2d 39 34 42 39 37 45 31 35 43 41 38   otgw-94B97E15CA8
01a0  33 2f 73 74 61 74 75 73 5f 73 6c 61 76 65 2d 43   3/status_slave-C
01b0  2d 46 2d 2d 2d 2d 30 27 00 22 4f 54 47 57 2f 76   -F----0'."OTGW/v
01c0  61 6c 75 65 2f 6f 74 67 77 2d 39 34 42 39 37 45   alue/otgw-94B97E
01d0  31 35 43 41 38 33 2f 66 61 75 6c 74 4f 46 46 30   15CA83/faultOFF0
01e0  2f 00 2b 4f 54 47 57 2f 76 61 6c 75 65 2f 6f 74   /.+OTGW/value/ot
01f0  67 77 2d 39 34 42 39 37 45 31 35 43 41 38 33 2f   gw-94B97E15CA83/
0200  63 65 6e 74 72 61 6c 68 65 61 74 69 6e 67 4f 4e   centralheatingON
0210  30 32 00 2d 4f 54 47 57 2f 76 61 6c 75 65 2f 6f   02.-OTGW/value/o
0220  74 67 77 2d 39 34 42 39 37 45 31 35 43 41 38 33   tgw-94B97E15CA83
0230  2f 64 6f 6d 65 73 74 69 63 68 6f 74 77 61 74 65   /domestichotwate
0240  72 4f 46 46 30 26 00 22 4f 54 47 57 2f 76         rOFF0&."OTGW/v
```
</details>

rhtenhove, 2025-01-16:
I had a similar idea that the ESP8266 was just being overloaded; however I've already disabled all sessions, gave the OTGW a power cycle, and made sure to not open any new session (such as telnet or the webbrowser), even temporarily, to ensure it is only connected to the MQTT broker.

Sadly this has been like that for 8 days and still the problems occurs.

I do see that every day there is a time block of several hours where the problem does not occur, but not the same time block every day. Also not a discernible pattern with people in the house, heat demand, or anything of the sort.

I'm going to try a different wifi channel, see what that does. I had already chosen the least occupied channel here (1), and made sure my zigbee network is on a non overlapping channel (25). But perhaps my neighbours do something odd most of the day, who knows.

rhtenhove, 2025-01-19:
I have tried a different channel (6 instead of 1) on my default Ziggo modem. I've also tried a different AP (Asus RT-AC68U). Still the same errors.

rhtenhove, 2025-01-30:
Further attempts which made no difference are:
- Greatly reducing the amount of MQTT messages sent
- Setting the timeout of pubsubclient to the default (15) instead of the set 4
- Using a different board (Wemos D1 mini Pro)

One thing that did change but not improve the behavior was to compile with lwip high bandwidth (`,ip=hb2f`), as from wireshark there seemed to be a pattern of breaking when a packet crossed the MSS threshold (not always). Now instead of mostly malformed packets, there's now various other errors popping up, including just closing the connection, protocol errors, timeouts.

```
2025-01-29 20:09:16: Client OTGW has exceeded timeout, disconnecting.
2025-01-29 20:09:58: New client connected from 192.168.178.95:50826 as OTGW (p2, c1, k15, u'esp32').
2025-01-29 20:10:33: Client OTGW has exceeded timeout, disconnecting.
2025-01-29 20:11:16: New client connected from 192.168.178.95:55549 as OTGW (p2, c1, k15, u'esp32').
2025-01-29 20:40:33: Client OTGW disconnected due to malformed packet.
2025-01-29 20:41:18: New client connected from 192.168.178.95:65388 as OTGW (p2, c1, k15, u'esp32').
2025-01-29 21:53:59: Client OTGW disconnected due to malformed packet.
2025-01-29 21:54:41: New client connected from 192.168.178.95:54769 as OTGW (p2, c1, k15, u'esp32').
2025-01-29 21:59:54: Client OTGW closed its connection.
2025-01-29 22:00:32: New client connected from 192.168.178.95:58681 as OTGW (p2, c1, k15, u'esp32').
2025-01-29 22:52:10: Client OTGW has exceeded timeout, disconnecting.
2025-01-29 22:52:52: New client connected from 192.168.178.95:58575 as OTGW (p2, c1, k15, u'esp32').
2025-01-29 23:03:47: Client OTGW already connected, closing old connection.
2025-01-29 23:03:47: New client connected from 192.168.178.95:60684 as OTGW (p2, c1, k15, u'esp32').
2025-01-29 23:04:22: Client OTGW disconnected.
2025-01-29 23:04:22: New client connected from 192.168.178.95:52419 as OTGW (p2, c1, k15, u'esp32').
2025-01-29 23:04:23: Client OTGW disconnected.
2025-01-29 23:04:23: New client connected from 192.168.178.95:56984 as OTGW (p2, c1, k15, u'esp32').
2025-01-29 23:05:40: Client OTGW has exceeded timeout, disconnecting.
2025-01-29 23:06:22: New client connected from 192.168.178.95:61803 as OTGW (p2, c1, k15, u'esp32').
2025-01-30 00:28:40: Client OTGW has exceeded timeout, disconnecting.
2025-01-30 00:29:14: New client connected from 192.168.178.95:49553 as OTGW (p2, c1, k15, u'esp32').
2025-01-30 01:01:52: Client OTGW has exceeded timeout, disconnecting.
2025-01-30 01:02:34: New client connected from 192.168.178.95:61968 as OTGW (p2, c1, k15, u'esp32').
2025-01-30 02:36:33: Client OTGW closed its connection.
2025-01-30 02:37:15: New client connected from 192.168.178.95:64099 as OTGW (p2, c1, k15, u'esp32').
2025-01-30 02:45:17: Client OTGW closed its connection.
2025-01-30 02:45:59: New client connected from 192.168.178.95:61715 as OTGW (p2, c1, k15, u'esp32').
2025-01-30 03:13:56: Client OTGW closed its connection.
2025-01-30 03:14:33: New client connected from 192.168.178.95:50911 as OTGW (p2, c1, k15, u'esp32').
2025-01-30 03:17:02: Client OTGW closed its connection.
2025-01-30 03:17:36: New client connected from 192.168.178.95:58819 as OTGW (p2, c1, k15, u'esp32').
2025-01-30 03:36:07: Client OTGW closed its connection.
2025-01-30 03:36:49: New client connected from 192.168.178.95:51941 as OTGW (p2, c1, k15, u'esp32').
2025-01-30 03:37:24: Client OTGW closed its connection.
2025-01-30 03:38:01: New client connected from 192.168.178.95:60715 as OTGW (p2, c1, k15, u'esp32').
2025-01-30 03:42:04: Client OTGW closed its connection.
2025-01-30 03:42:46: New client connected from 192.168.178.95:64747 as OTGW (p2, c1, k15, u'esp32').
2025-01-30 04:00:40: Client OTGW has exceeded timeout, disconnecting.
2025-01-30 04:01:22: New client connected from 192.168.178.95:60432 as OTGW (p2, c1, k15, u'esp32').
2025-01-30 04:05:05: Client OTGW disconnected due to protocol error.
2025-01-30 04:05:47: New client connected from 192.168.178.95:54636 as OTGW (p2, c1, k15, u'esp32').
2025-01-30 05:05:14: Bad client OTGW sending multiple CONNECT messages.
2025-01-30 05:05:14: Client OTGW disconnected due to protocol error.
2025-01-30 05:05:56: New client connected from 192.168.178.95:50597 as OTGW (p2, c1, k15, u'esp32').
2025-01-30 05:11:40: Client OTGW has exceeded timeout, disconnecting.
2025-01-30 05:12:22: New client connected from 192.168.178.95:50173 as OTGW (p2, c1, k15, u'esp32').
2025-01-30 05:24:46: Client OTGW has exceeded timeout, disconnecting.
2025-01-30 05:25:30: New client connected from 192.168.178.95:62945 as OTGW (p2, c1, k15, u'esp32').
2025-01-30 05:31:16: Client OTGW disconnected due to malformed packet.
2025-01-30 05:32:00: New client connected from 192.168.178.95:63391 as OTGW (p2, c1, k15, u'esp32').
2025-01-30 06:36:10: Client OTGW has exceeded timeout, disconnecting.
2025-01-30 06:36:52: New client connected from 192.168.178.95:52217 as OTGW (p2, c1, k15, u'esp32').
2025-01-30 06:39:00: Client OTGW closed its connection.
2025-01-30 06:39:42: New client connected from 192.168.178.95:63526 as OTGW (p2, c1, k15, u'esp32').
2025-01-30 06:43:48: Client OTGW disconnected due to malformed packet.
2025-01-30 06:44:30: New client connected from 192.168.178.95:61309 as OTGW (p2, c1, k15, u'esp32').
2025-01-30 06:45:19: Client OTGW closed its connection.
2025-01-30 06:46:01: New client connected from 192.168.178.95:50273 as OTGW (p2, c1, k15, u'esp32').
[... cut ...]
2025-01-30 07:30:28: Client OTGW has exceeded timeout, disconnecting.
2025-01-30 07:31:10: New client connected from 192.168.178.95:55038 as OTGW (p2, c1, k15, u'esp32').
2025-01-30 07:46:33: Client OTGW closed its connection.
2025-01-30 07:47:15: New client connected from 192.168.178.95:53531 as OTGW (p2, c1, k15, u'esp32').
2025-01-30 08:09:14: Client OTGW closed its connection.
2025-01-30 08:09:53: New client connected from 192.168.178.95:50225 as OTGW (p2, c1, k15, u'esp32').
```

Still no solution. TCP seems to be having a real hard time here.

critictidier, 2025-02-10:
Similar issues here. There are days like today where the connection is very unstable.

[image]

bmxbandita, 2025-02-27:
Similar issues here as well. Fresh out of the box, soldered and flashed by nodo shop, all i did was to configure the wi-fi network.

DaveDavenport, 2025-02-27:
For me I have no issue in the data I get back from the board.
Given it seems to be split, maybe we can find some 'common' factor between the people that have an issue.

* Nodo Shop Board version.
**  (will check when I get home)
* ESP wemos board version. 
   (I have received boards before from tinytronics that had a very slow flash chip (Puya p25q32sh), that caused very odd issues. So maybe it is something like this)
** Wemos D1 mini v3.0.0
* What AP is used.
** TP-link Omada TP-Link EAP653
* Uptime of the wemos (I wonder if the wemos resets itself?)
** 21 days (was a power cycle I did).

bmxbandita, 2025-03-02:
@DaveDavenport 
Turns out a bad wifi signal was the culprit in my case.
Relocated the OTGW, now its steady as a rock!

ronajon, 2025-04-04:
same issue here. following!