rvdbreemen/OTGW-firmware #266: ESP not connecting to wifi

Basb1234, 2025-07-05:
> I have exactly the same problem as ministryofsillywalks. I tried a different ESP, but still the same problem. Both ESP were successfully flashed with the latest firmware.
> I bought a new router, but still the same issue. 
> Are there certain requirements that the network should comply to? e.g. only 2,4 Ghz, WPA2, ...  

 _Originally posted by @Basb1234 in [#248](https://github.com/rvdbreemen/OTGW-firmware/issues/248#issuecomment-3017098652)_

Roos-AID, 2025-07-05:
All,Many ESPs suffer from power issues when connecting to Wifi. Please make sure tou have proper powersupply and if possible measure the 3v3 on the ESP when connecting. On many chinese clone ESP I found significant drop of voltage when the wifi kicks in. This is causing the connection to fail. Met vriendelijke groet / Best regardsRob RoosVerstuurd vanaf mijn iPhone

Basb1234, 2025-07-05:
Hi Roos-AID,
Thanks for the fast reply. The ESP is **not** getting back into AP mode. Issue https://github.com/rvdbreemen/OTGW-firmware/issues/100 suggests that it might have a connection. I went to http://otgw.local and saw a website :)
I have tried this before on my previous ESP, even though the router didn't show the device in the list of connected clients, but without luck. 
I now have a more fancy router with a IoT network. I had a 2.4 Ghz guest network on my previous router (ASUS RT-AX88U), but then I wasn't able to reach http://otgw.local. I also tried it with disabling the 5Ghz network and connecting it on my main network on my previous router.

Still, for some reason, I don't see the device in the client list on my new router (Asus RT-BE88U) while I am able to access http://otgw.local. Is that normal ?

Roos-AID, 2025-07-06:
Can you make screenshot of the screen that is shown when you access otgw.local ?  

Couple of questions
1. Can you initiate a Telnet session(port 23)  to the IP address of the ESP ? 
2. Do you have a NTP server on your network ?  Can it be that NTP requests are not answered fast enough by your network ? This might cause the ESP to hang getting NTP time, it can take up to couple of minutes , did you wait that long ? 
3. Next to flashing the firmware, did you also flash the Filesystem with fs.bin ?

Roos-AID, 2025-07-07:
Well, then all seems to working as it should.  I don't have a clue why you cannot find the IP on the router. 
I would recommend to set a fixed IP (reserved IP)  in DHCP for the OTGW.

On the Web page, you can select Advanced and then  Debug Information.
You will see there the Wifi information such as Rx signal quality. 

Good luck with next steps.

closed 2025-12-07 by rvdbreemen